portapack-mayhem/mayhem-firmware #1061: Support for viewing Screen-Shots

NotherNgineer, 2023-05-24:
Low priority, but it would be nice if there was a way to view screen-shots without having to use a computer.

Implementation-wise, I see there is code for viewing a BMP file, but screen-shots are saved in PNG format.  Perhaps it would be easiest to change the function that saves screen-shots to use BMP format, TBD.

BTW, I am not suggesting that we implement support viewing files with different formats or resolutions.  My intended use case was just for viewing screen-shots.

Brumi-2021, 2023-05-24:
Hello, good idea, @NotherNgineer , and  good job !!!!  yes I was also missing some stand alone ,  screen-shots viewer.
When I was checking if we could control the backlight dimming (it seems yes but only in H1, but I made several test and I could not success on it ) .  Anyway , I also could see some  comments about BMP implementation  in that file , /opt/portapack-mayhem/firmware/common/lcd_ili9341.cpp
[image]

But I do not know know if those comments are still valid or not ??? I did not not study it .
Cheers,

NotherNgineer, 2023-05-24:
@Brumi-2021 thank you for the info.

@kallanreed also mentioned viewing screen-shots without a computer and may have some thoughts on this as well.

kallanreed, 2023-05-24:
An uncompressed, 240x320*24bpp should be about 230kB and IIRC that’s about what the PNG files are?! Are they uncompressed then?

BMP would certainly be less code to draw. I haven’t looked too hard into PNG format to know if it’s memory expensive to decompress them. That’ll be the trick - I was hitting OOM at 44kB.

kallanreed, 2023-05-24:
So yeah, the PNG files are uncompressed. Might as well output a BMP file then. Much simpler to read and write, IMO.

```
static constexpr std::array<uint8_t, 25> png_ihdr_screen_capture{{
    0x00, 0x00, 0x00, 0x0d,  // IHDR length
    0x49, 0x48, 0x44, 0x52,  // IHDR type
    0x00, 0x00, 0x00, 0xf0,  // width = 240
    0x00, 0x00, 0x01, 0x40,  // height = 320
    0x08,                    // bit_depth = 8
    0x02,                    // color_type = 2
    0x00,                    // compression_method = 0
    0x00,                    // filter_method = 0
    0x00,                    // interlace_method = 0
    0x0d, 0x8a, 0x66, 0x04,  // CRC
}};```

NotherNgineer, 2023-06-04:
Resolved by PR #1102

Thank you @kallanreed

closed 2023-06-04 by NotherNgineer