portapack-mayhem/mayhem-firmware #1063: Bad touch behavior of bottom-left corner

zxkmm, 2023-05-24:
**Describe the bug**
Touch is not sensitive for bottom-left corner  
(It's **not** related to @bernd-herzog 's multiple touch fix)  

**Expected behavior**
Touch detected

**Affected versions**
since over  
(It's **not** related to Bernd's multiple touch fix, cause it has the problem since over)  

**Additional**
I took a look at the code, and considering its behavior, it mostly looks like it's caused by bad radius calculation of `calculate_metrics` but I couldn't figure out what's wrong of it.  
And I did some tests, it seems the top-right corner is most sensitive, and the bottom-left corner is the least sensitive, Happend on both H1 and H2, but since H2 has larger screen, it's more serious on H2. 
So I wrote a curve function to compensate for the loss of radius, but not working, I tried `log10x + 2` and `10x` and `x^2`, they all have different behavior but nothing works.  

From what I know, usually a `4-wire resistive touch screen` should pass through an `ADC`, but I took a look of circuit of H1 and H2, it seems it directly go through the `5m40z`, but I think it's still possible to fix by code, but I don't know how.  
It would be really nice if someone could figure out and fix.    
Thank you so much in advance!!

vmakeev, 2023-05-25:
Same as here #328

eried, 2023-05-25:
Yeah as @vmakeev said, duplicated of https://github.com/eried/portapack-mayhem/issues/328

zxkmm, 2023-05-26:
@eried not hardware issue. fixed. test binary: https://discord.com/channels/719669764804444213/722101917135798312/1111495356023320648