portapack-mayhem/mayhem-firmware #1065: SD over USB issue?

sizzlik, 2023-05-24:
Hi, i got an HackRF r9 clone and PortaPack H2 from OpenSourceSDRLab and for some reason the SD over USB feature wont work for me. Im on Win10 and DragonOS Focal, the firmware is updated to the latest (n_230524 but the sd/usb feature didnt work before either.)
I tried different cables, different usb ports, made sure the cable is shielded with a multimeter, tried using an active usb extension, but neither on windows nor DragonOS anything happens when i start the feature. Is there a trick or a known problem?

jLynx, 2023-05-25:
https://github.com/eried/portapack-mayhem/wiki/SD-card-not-recognized-by-PC-with-the-SD-card-over-USB-selected

vmakeev, 2023-05-25:
Same problem. Unfortunately, I cannot follow the instructions from the answer above, because no new entries appear in the windows device manager when SD over USB mode is turned on (when switching to HackRF mode, a new device appears, but this does not give access to the SD card)

u-foka, 2023-05-27:
Should be fixed by PR #1081, test binary can be found [here](https://discord.com/channels/719669764804444213/722101917135798312/1111770798097698896).

jLynx, 2023-05-28:
Closing this issue as it should now be fixed. If you are still facing it, let me know and we can re open this

closed 2023-06-12 by jLynx