portapack-mayhem/mayhem-firmware #1066: Persistent "Audio Level" setting

NotherNgineer, 2023-05-25:
Every time I turn on my Portapack H2+, I have to increase the volume from the default of 50 to a level that I can actually hear.

I was thinking that we could store the last audio level in persistent memory, but I don't know if we'd want to re-write persistent memory that often.

So, alternatively, could I perhaps add a "Default Audio Volume Level" setting to the existing Settings->Audio configuration page (saving that setting in persistent memory of course)?

This would be a really simple change, and could save some of us users from having to turn up the audio level every time they power up the device.

Thoughts?

(If it sounds like a good idea, please assign to me)

kallanreed, 2023-05-25:
Go for it 👍 
I'm the opposite, I always have to turn it down for my headphones.

eried, 2023-05-25:
Maybe this should be discussed in discord, since involves a small change in every app. I am not really sure if custom default is better than keep last value. 🤔

gullradriel, 2023-05-25:
I vote keep last value per app

Brumi-2021, 2023-05-25:
Hello  I have same feeling as @NotherNgineer , when I needed , I always needs to increase 50 to 60 or 70 .
To me , also "keep last value" , is enough .

NotherNgineer, 2023-05-25:
I'm now thinking that receiver_model.set_headphone_volume() could save the last configured volume setting in persistent memory (this is called from most apps, so no need to modify every app).  No need to add a new setting to the Settings->Audio page.

This would be similar to the receiver_model.set_tuned_frequency() function which consistently updates persistent memory to save the last tuned frequency.

I would definitely recommend against remembering volume level on a per-app basis.

(I am also curious why some apps such as AFSK and APRS have no way to actually configure the volume level, just relying on the last volume level configured by the previous app)

kallanreed, 2023-06-06:
Fixed in #1110

closed 2023-06-06 by kallanreed