portapack-mayhem/mayhem-firmware #1067: C16 File Viewer / Extractor app

NotherNgineer, 2023-05-25:
Similarly to the WAV file viewer app, I think it would be nice to have a viewer app for C16 capture files without requiring a computer.

Ideally it should be able to crop them, and extract/save only the important signals to separate C16 files (either manually using start & end markers, or automatically based on signal level).

I realize Portapack is memory-limited and only a portion of the file could be loaded into RAM at a time (the app would need to do a lot of SD Card I/O while scrolling through the data).  Also note that I'm not suggesting any sophisticated signal processing transformations at this time; just a viewer and simple chop-editing tool.

This would help with the Replay app, by not having to transmit the entire capture file.

NotherNgineer, 2023-05-29:
BTW, I have used Audacity to do this, but:
1)  Audacity requires a computer, which defeats the whole "Porta" part.
2)  The Audacity method requires a few more steps than would be necessary with an app.

eried, 2023-05-29:
Split/trim might be possible, joining might be more tricky because possible differences on the capture.

eried, 2023-05-29:
Consider that you can promote your own suggestion by adding a bounty on it https://github.com/eried/portapack-mayhem/#what-if-i-really-want-something-specific

NotherNgineer, 2023-10-08:
Trim feature implemented in #1456 & #1458 - thanks @kallanreed