portapack-mayhem/mayhem-firmware #1070: faulty GPS signal for R9 version?

Robert19721106, 2023-05-25:
I have an older version of HackRF One (with MAX2387 chip) and two new R9 cards (with MAX2839 chip) and one PortaPack H1 card (from older HackRF One version)

After uploading the mayhem firmware to new boards (R9) with PortaPack connected, the GPS simulator does not work properly - the signal is "narrowband" - attached is a screenshot from the spectrum analyzer.

[image]

After uploading the same firmware to the older version of HackRF One (the one based on MAX2837), everything is OK - a screenshot of the emitted GPS signal is attached.

[image]

Is the problem caused by a bug in the software (incorrect MAX2839 support?) or is my PortaPack H1 board not compatible with the newer R9 version ?

Regards,
Robert

Brumi-2021, 2023-05-25:
Hackrf_info la detecting it as hw r9 ?

Robert19721106, 2023-05-25:
I checked all boards under microscope and these two new boards have MAX2389 instead of MAX2387 (as my old board).
So this is R9 version. I have old executables of hackrf_info , so answer is not completed: 

Found HackRF board.
Board ID Number: 4 (Unknown Board ID)
Firmware Version: 2023.01.1
Part ID Number: 0x0054435d 0x0054435d
Serial Number: 0x00000000 0x00000000 0x118461dc 0x21249b03

If you send me newer hackrf_info.exe (for Windows) I can check these new board, but for sure it contain MAX2839.

I tested portapack-mayhem V1.7.0 and the newest (V1.7.1?) and portapack-havoc-modified (https://github.com/jamesshao8/portapack-havoc-modified) and results are the same...

BTW: GPS signal generation in HACKRF mode are OK for each board (https://github.com/osqzss/gps-sdr-sim) but GPS signal generation in standalone mode (with PortaPack) is wrong for hackrf one with MAX2839 chip.

Brumi-2021, 2023-05-25:
Hello Robert, I do not have a r9 .
But I could see other colleagues with r9 ,
And in that case Hackrf_info is detecting it 

I just delete his serial number . 

```hackrf_info version: 2022.09.1
libhackrf version: 2022.09.1 (0.7)
Found HackRF
Index: 0
Serial number: 0000000000000000xxxxxxxxxxxxxxxx
Board ID Number: 4 (HackRF One)
Firmware Version: v1.7.1 (API:1.07)
Part ID Number: 0xa000cb3c 0x00xxxxxx
Hardware Revision: r9
Hardware does not appear to have been manufactured by Great Scott Gadgets.
Hardware supported by installed firmware:
    HackRF One```

I believe that he just used the hackrf_info from the zip distribution  . 

Did you used a recent version of Hackrf_info ?

Brumi-2021, 2023-05-25:
I think , first we need to achieve that your board is recognised as r9 by hackrf_info .

Just In case that in our Mayhem releases we still do not have updated the new tools for r9 , better that you download it from here 
 try the latest hackrf bin. download 2023.1.1 here: 

https://github.com/greatscottgadgets/hackrf/releases

You will find a hackrf_one_usb.bin in firmware-bin in that zip.  If you are working on windows: place it where your dfu_hackrf_one.bat is and do the dfu flash again, that worked once . And also you need  the hackrf_one_usb.dfu you need to  replace both . 

Then from DFU, write first the hackrf_one_usb.dfu  ,
And then check hackrf_info . Once detected r9 , you can overwrite the latest Mayhem fw 1.70 or 1.71 or latest nightly.

Robert19721106, 2023-05-26:
I can not find the newest hackrf_info.exe (binary for Windows). 

BTW: https://hackrf.readthedocs.io/en/latest/installing_hackrf_software.html ---> but Windows binaries provided as part of the PothosSDR are from 2021... (but R9 board is made from 2023)

Can you send me hackrf_info.exe which recognize R9? (I have not time for compilation it from sources).

Brumi-2021, 2023-05-26:
Sorry Robert,
 i also have not that latest hackrf_info  file in windows . 

I try to follow all Hackrf issues in linux .
You are right, PhotosSDR build is from 2021 !!! (Not latest one )  maybe it would be easier if you can  access to any other laptop with linux, is it possible ?

Robert19721106, 2023-05-26:
I have UBUNTU @ Windows10, so it should be OK, can you send me binaries of hackrf tools for Linux?

Brumi-2021, 2023-05-26:
Hi , 
I have pentoo (gentoo) distro , so I am afraid that I can Not transfer you my local Hackrf tools compiled as binaries ebuilds ,  to your different linux distro . 

I think , the easy way , follow Hackrf Ubuntu install instructions . If need we can ask help to other Ubuntu colleagues in Discord . 

Cheers

Robert19721106, 2023-05-27:
Hi Brumi!

My new board:

Found HackRF
Index: 0
Serial number: 0000000000000000118461dc21935803
Board ID Number: 4 (HackRF One)
Firmware Version: 2023.01.1 (API:1.07)
Part ID Number: 0xa000cb3c 0x0061475e
Hardware Revision: r9
Hardware does not appear to have been manufactured by Great Scott Gadgets.
Hardware supported by installed firmware:
    HackRF One

My old board:

Found HackRF
Index: 0
Serial number: 0000000000000000088869dc35994c1b
Board ID Number: 2 (HackRF One)
Firmware Version: 2023.01.1 (API:1.07)
Part ID Number: 0xa000cb3c 0x00544f58
Hardware Revision: older than r6
Hardware supported by installed firmware:
    HackRF One

It seems that support for MAX2839 contains some important bug. 

To sum up, the GPS simulator does not work properly for the R9 version - it emits some strange narrow-band signal instead of a typical wide-spectrum signal, earlier I sent screenshots from the spectrum analyzer for both versions of HackRF One and the same firmware version.

Robert

Brumi-2021, 2023-05-27:
Ok good , how about the rest of community members that has HackRF r9 ? Can someone else try to play GPS Simul and see with other SDR Receiver if the radiated spectrum is narrow (as described by Robert ) one instead of wide spectrum (normal ) ?

u-foka, 2023-05-27:
I've tested with r9 + H2 from ali, receiving with an RTL2832 based DVB tuner, the signal looks alright, anyhow I change gain / amp settings.
[image]

If I find my old android phone, will test with that too..

Brumi-2021, 2023-05-27:
Hi @Robert19721106 , as you could see other colleague with r9 , transmit a correct wide broadband using GPS Simul and the default .C8 example files . 
Then it seems more a problem to your board .
Other TX modulation goes well , WFM , NFM , AM , SSB . morse ?

Robert19721106, 2023-05-27:
I checked the transmitter/sound broadcasting mode and found something interesting to track down the bug.

Well, there is an unmodulated carrier there, but after a few START/STOP it starts working (I can hear the WAV file being broadcast in the external FM radio receiver and I can change the FM deviation value and it causes the correct volume change in the receiver). After this "unblocking" when I enter the GPS simulator mode, the transmitted signal is correct ...
And it works properly until RESET / power-up of HackRF One...

This is some sort of software or hardware hazard.

So there can only be two explanations:

option 1) I have both defective newly purchased HackRF One R9 tiles (e.g. incorrect value of some electronic component), tiles purchased on Ali, Chinese of course, but I would not prejudge the fault

option 2) there is a glitch in the initialization of the radio path (e.g. the MAX2839 initialization procedure) which results in an uncertain (hazardous) result, e.g. an unlocked PLL sync loop or something like that, which causes that the modulation does not work; after a few manipulations START/STOP for different modes of the transmitter "something" and then everything works fine (until next power-up)

Any ideas?

Brumi-2021, 2023-05-27:
Hello  , I just have this idea , 
 at the moment , in the past ,  we have found several colleagues having some strange  modulation problems , and they were due to have a very old  Hackrf Xilinx  CPLD firmware not updated. 

So far we have found two clear ways to detect indirectly if you have old CPLD , 
(1) Check #1009 
(2) Check #651 

Maybe the easy one for you would be check (1) . 
Let me know if you could recognise any of those (1) , (2) problems in your new Hackrf r9 , 
and if yes,  I advice you to update to try to update the Hackrf CPLD fw , following the the (1) description, 
and then repeat the test of  GPS Simul ,and other Modulation incidences.

Cheers.

Robert19721106, 2023-05-28:
Hi Brumi!

In my case, autoupdating the CPLD by installing the latest hackrf_one firmware didn't work (tested on both R9 boards), while manually reprogramming the CPLD ( ./hackrf_cpldjtag -x default.xsvf ) yielded positive results and both R9 boards now work fine. 

Thank you for your help.

Cheers.

Brumi-2021, 2023-05-28:
Hi @Robert19721106 , that it is excellent results !
We are really happy about it .👍🎉🎉🎉🎉
Just before closing that issue , could you let me know if you could identify it with the checking (1) and (2) 

cheers

Robert19721106, 2023-05-28:
it was (1):

[image]

Brumi-2021, 2023-05-28:
Ok thanks @Robert19721106 to give us full details .
We take note about it  , to advice to other users with similar problem , to go straight away to the direct CPLD flash methode , and not rely about the automatic one . 

I guess that now , you also fixed all other modulation incidences .  Then let’s  gladly close that issue !!! 
cheers

closed 2023-05-28 by Brumi-2021

Brumi-2021, 2023-05-28:
BTW @Robert19721106 , as you could probable see . 
Latest nightly Mayhem binary , also  solves the #1081 (that also was  related  to  your  Hackrf version r9 )
(Not included yet  in latest official fw v 1.70, nor 1.71)

Cheers

Sats301, 2023-07-25:
Hello everybody!

Completely same situation, but with one little differens...
1. All actions with FW/CPLD manually flashing - not working.
2. Can't check situation with 1.6.0 and lower FW (screen not working, buttons can't help).
3. I've bought 100 PCS of HackRF PortaPack H2+ for student education, and the part of education program - not available, because we cant discover lite spoofing...
4. .C8 folder generated by gps-sdr-sim pack.

I am really not understand what I need to do.
Just look on photos...
[image]
Hacrf_info

[image]
Chipsets

[image]
GPS_Sim - completely bad!
[image]

Friends could I ask you for a help?
Because this situation is a really nervous...