portapack-mayhem/mayhem-firmware #1074: Signal editing after saving

utenus, 2023-05-26:
**Describe the solution you'd like**
It would be extremely useful during security tests if the recorded signal could be edited on PortaPack. Although some basic editing options as it is, for example, in Universal Radio Hacker. In addition, viewing the signal as binary values ​​would also be useful.

Commander-Crash, 2023-05-29:
Audacity https://github-wiki-see.page/m/eried/portapack-mayhem/wiki/C16-format

sizzlik, 2023-06-03:
Yes.."simple" cut/crop functionality to create signal snippets would be quite awesome. Most other stuff is probably better done on pc, but croping out the wanted 0,5sec signal from a several second recording, or cuttiung up a recording that contains 3 different signal bursts into 3 different recordings, on the go, very welcomed feature.

NotherNgineer, 2023-10-08:
Trim feature implemented in #1456 & #1458 - thanks @kallanreed

gullradriel, 2024-04-08:
I consider this to be advanced enough for the hardware.

closed 2024-04-08 by gullradriel