portapack-mayhem/mayhem-firmware #1077: Looking glass waterfall is inaccurate

vmakeev, 2023-05-26:
After upgrading from 1.7.0 to 1.7.1, I found that the waterfall display looks incorrect in Looking Glass: the frequencies are shifted.

For the experiment, I took a frequency range of 100-110 MHz, in which the station with a frequency of 106.6 MHz has the strongest signal, and set the marker to this frequency.

In firmware version 1.7.0, the marker relatively accurately indicates the display of this station on the waterfall. 

[image]

In version 1.7.1, the marker is located in the same place, but the signal of the waterfall station itself is shifted to the side by about 2 MHz.

[image]

The last night build (n_230526) has the same problem

gullradriel, 2023-05-26:
I will take a look

vmakeev, 2023-05-26:
One more observation: if the range is <20MHz, then DC spike is clearly visible on the waterfall, if 20 or more, then it is not visible. That's by design?

19 MHz range:
[image]

20 MHz range:
[image]

gullradriel, 2023-05-26:
Under 20 MHz it's made in one pass so DC spike is here.
After 20MHz it is no more in one pass so the DC spike is ignored in Fast mode, and it is avoided in slow mode.

gullradriel, 2023-05-26:
@vmakeev please test the bin here: https://discord.com/channels/719669764804444213/722101917135798312/1111710046305407077
And tell me if it solves your problem :-)

vmakeev, 2023-05-26:
First of all, thank you so much for your work, it has become noticeably better. 

Nevertheless, a certain inaccuracy remains. I'll try to describe what I'm observing.

**Case 1:** 
* bandwidth = 20 MHz (slow mode)
* radio station frequency = 106.6 MHz

The marker is set to 106.6667 MHz (I can't set it more precisely in Looking Glass)

The radio station is displayed on the waterfall to the left of the marker

[image]

If I move the marker exactly to the radio station display, then its value value will be 106.2 MHz (-400 kHz relative to the real frequency)
[image]

Next, I switched from Looking Glass to Audio RX at the moment when the LG marker was set to display the radio station (106.1667 MHz). The waterfall in Audio RX is empty at this frequency, the radio station is displayed in the correct place on the waterfall (slightly to the right, at 106.6 MHz)

[image]

**Case 2:**

* bandwidth = 19 MHz (fast mode)
* radio station frequency = 106.6 MHz

The marker is set to 106.6578 MHz (I can't set it more precisely in Looking Glass)

The radio station is displayed on the waterfall to the left of the marker

[image]

If I move the marker exactly to the radio station display, then its value value will be 106 MHz (-600 kHz relative to the real frequency)

[image]

gullradriel, 2023-05-29:
I made https://github.com/eried/portapack-mayhem/pull/1091 which is a slighlty better version than the last one.
It would be nice if you can test that one which take the last things mentioned in account:
https://discord.com/channels/719669764804444213/722101917135798312/1112670768732573767
Many thanks !

vmakeev, 2023-05-29:
In the case of FM radio, the accuracy has improved, but the sensitivity has significantly decreased (it is difficult to visually find the frequency of the station on the waterfall).

I also checked on a keychain with a frequency of ~ 316.5 MHz, the results are below:

Audio RX SPEC - precise
[image]

Looking Glass (latest from Discord): the signals are somewhere far to the right of the marker
[image]

I'm afraid that the version of the application from 1.7.0 works much more accurately and predictably than the new one:(

Looking Glass from 1.7.0:
[image]

gullradriel, 2023-05-30:
Thanks for your tests. I made a new test bin in test drive.
In the last bin you tested I made a bad copy paste on the spectrum bandwidth set.
It s fixed in my version.
I have a few things to check and I'll update this and the test drive.

gullradriel, 2023-05-30:
In case you want to test before I'm making more tuning, here you go: https://discord.com/channels/719669764804444213/722101917135798312/1112969629762068480

vmakeev, 2023-05-30:
The signals on the waterfall have become much more visible, thanks!

As for accuracy, the keychain with a frequency of 316.5 MHz now matches the marker fairly accurately.

With the radio station, the accuracy returned approximately to the values of this message https://github.com/eried/portapack-mayhem/issues/1077#issuecomment-1564911713

----- 
Latest from Discord (e50d8dc1):

Slow mode:
[image]

Fast mode:
[image]

gullradriel, 2023-05-31:
Hi back @vmakeev and sorry for the delay. That's life, you know ^^
Anyway I have a good news. I think I fixed it !
Please try the bin here: https://discord.com/channels/719669764804444213/722101917135798312/1113565585742245929

I had pretty cool results with known FM stations in all modes, and with my curtains keyfob.

gullradriel, 2023-05-31:
Took an example like yours and here is the result:
Expected FM stations at 92.0 and 93.0, picture is 92.0:
[image]
The little margin you can see is due to the low resolution and calculus that is derived from the screen width, so it's acceptable now :-)

gullradriel, 2023-05-31:
868.975 signal picked correctly at 868.8353 (marker position). There is a lots of reflexions, but it's at the good place, since the next marker position is at 869.052.
[image]

gullradriel, 2023-05-31:
BTW I changed the limit so we can do up to 20 MHz included single pass instead of up to 19.

vmakeev, 2023-06-01:
Brilliant!

I checked the accuracy of the marker positioning in the FM band (both in fast and slow modes, 19 and 21 bandwidth), and on the keychain (316 MHz), and on the entire Wi-Fi/BT band (2.4 GHz). The positioning accuracy is amazing (+- one pixel)

@gullradriel, I am very grateful to you for your work, my hat is off to you, thanks a lot :)

gullradriel, 2023-06-01:
Thank you so much for your patience and all the tests you did for the validation / problem research !
I'm happy to close the issue  as well as finally merging the patch Pull Request. It will be available in today's nightly for all.
Your hat may be off, but my thumb is up to you and the few discord users who helped into the tests too :-)