portapack-mayhem/mayhem-firmware #1087: Remotely-controlled Portapack

NotherNgineer, 2023-05-28:
Consider, if you will, remotely operating one Portapack over-the-air using a different Portapack.

A protocol would need to be defined to send commands (Dial Turns, Button, and Touchscreen Presses) over the air to the secondary device, and acknowledge receipt.  Basic information could be sent back to the primary device such as log files, but nothing fancy (no "Waterfall" displays).  HackRF cannot transmit and receive at the same time, so both devices transmitting at the same time would result in neither receiving anything (randomly-timed retries might solve that).

Just because something may be "possible" doesn't make it useful, or worth doing.  While I try to imagine an application for this, I thought I'd just through the idea out there to see if there's any interest.  :-)

gullradriel, 2023-05-29:
I don't think it can work since you can not 'control it' and use the apps at the same time.
You would need a separated channel communication, or only use non radio apps, which make the demande pretty useless.
If you really want to develop something useful, a simple file transfert over radio would be great ^^

NotherNgineer, 2023-05-29:
I'm just going to close this enhancement suggestion since it doesn't seem practical or needed at the moment.

closed 2023-05-29 by NotherNgineer