portapack-mayhem/mayhem-firmware #1088: 1.71 is not saving recordings. it does act like is recording file normally but you can not find it on sd card . the file count (while recording) is growing tho. I have downgraded to 1.70 and all works fine !

DavidIgor, 2023-05-28:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly version here: https://github.com/eried/portapack-mayhem/releases)

----

**Describe the bug**
A clear and concise description of what the bug is. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Tap on '....'
3. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release:
* Latest Nightly release:
* Previous working release:

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.

jLynx, 2023-05-28:
Closed as issue is not formatted correctly

closed 2023-05-28 by jLynx

gullradriel, 2023-05-29:
Just in case the OP comes back here: the records and logs are not located at the root of the sdcard anymore. Please check the LOGS/, CAPTURE/ and AUDIO/ directories.