portapack-mayhem/mayhem-firmware #1089: problem enter HACKRF ONE mode after flashing nightly release https://github.com/eried/portapack-mayhem/releases/tag/nightly-tag-2023-05-27

apple601601601, 2023-05-28:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly version here: https://github.com/eried/portapack-mayhem/releases)

----

**Describe the bug**
Using hardware H2 R4: https://github.com/eried/portapack-mayhem/wiki/PortaPack-Versions#h2r4 
Using Firmware: https://github.com/eried/portapack-mayhem/releases/tag/nightly-tag-2023-05-27 
PC Operating system: Windows 11

Problem: Didn't see HACKRF ONE usb device in 'Device Manager' after clicking 'Hackrf one' mode icon and press 'YES'. 

**To Reproduce**
Steps to reproduce the behavior:
1. flash in nightly release: https://github.com/eried/portapack-mayhem/releases/tag/nightly-tag-2023-05-27
2. Plugin USB cable and connect to PC.
3. Tap on 'Hackrf one' mode icon and press 'YES' in the following screen
4. Refreshed 'Device Manager' and didn't see any new USB device appears.

**Expected behavior**
Hackrf one appears in 'Device Manager'

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release: https://github.com/eried/portapack-mayhem/releases/tag/nightly-tag-2023-05-27 
* Latest Nightly release: https://github.com/eried/portapack-mayhem/releases/tag/nightly-tag-2023-05-27 
* Previous working release: https://github.com/eried/portapack-mayhem/releases/tag/v1.7.1

**Additional**
I flashed back to v1.7.1 in DFU mode and it worked fine now. Not sure if related to specific hardware version I'm using: H2 R4.
Didn't try anyversion between v1.7.1 and nightly-tag-2023-05-07 though. 
(Plush SD over USB doesn't work in either version.)

jLynx, 2023-05-28:
Please try https://github.com/eried/portapack-mayhem/releases/tag/nightly-tag-2023-05-28 as that should fix sd over usb

apple601601601, 2023-05-28:
> Please try https://github.com/eried/portapack-mayhem/releases/tag/nightly-tag-2023-05-28 as that should fix sd over usb

Tried 230528, fixed both HACKRF mode and SD over USB issue. Thanks!

closed 2023-05-28 by apple601601601