portapack-mayhem/mayhem-firmware #1099: h1 speaker hide falling to show speaker icon

keagan73, 2023-06-03:
once the batt is dead it will no longer show the speaker icon.
even if i use the "p memory mgmt" and re save, reset than save. i cant get it to show the speaker icon by default. 
the only way iv been able to get it to come back is disabling it and re enable it.

gullradriel, 2023-06-04:
It's not enough to just use save. The Pmem mgmt is just here as a helper for those without coin battery and needs a sdcard to function.
In short:
-each time you make a modification you want to save, you explicitly have to go back in pmem settings and click 'save'
-the 'load pmem at startup' have to be checked

gullradriel, 2023-06-04:
I removed my previous message because you're right. Something is messy with the speaker icon. It looks like it's loaded before pmem settings are restored.

gullradriel, 2023-06-04:
I pushed a PR solving the problem. You can try the bin based on next + the fix here:
https://discord.com/channels/719669764804444213/722101917135798312/1114921059045617674

closed 2023-06-05 by gullradriel