portapack-mayhem/mayhem-firmware #1100: H1 Audio on H2+

sizzlik, 2023-06-03:
Hi,
this might be more observation+question, than bug report:
I just installed a small laptop speaker in my toybox (hRF one r9 + PP H2+). Works fine when recieving audio on FM band. So i switched on the option to show H1 audio switch and it works, mutes the audio.
Now the observation i made is that, no matter the initial state when turning on the device, Audio will be on.
I saved settings, so that when turning the device on, the Icon is green, works as expected.
I saved settings, so the icon is grey when turning on, the audio is still turned on. So you have to switch from grey to green back to grey to turn off. Maybe that cannot be changed, but would be nice to start the device with sound disabled.
And additional question: What does the H1 Key tone option do thats set to 20%?

(And thanks for fixing the SD-card issue)

gullradriel, 2023-06-04:
Do you have a coin battery installed ? Is it working ? Do the clock stays up to date ?
Are you using pmemsettings ?

sizzlik, 2023-06-04:
Yes to all. Battery installed (and charged), PP keeps time when i remove the big battery. Keeping the settings is no issue. As said, when i save the settings that the audio icon is grey on startup, it is still grey on startup but the audio is not muted.

gullradriel, 2023-06-05:
Well. I don't know if we can do something for that.
On a side note, if your coin battery is working you should not use what's under Settings/PMem Mgmt  (maybe you're not, it's just a precision).

NotherNgineer, 2023-06-06:
I'm not sure of the expected behavior, but on my H2+ I observe (using latest nightly) that if the H1 Speaker Option is set to "show" the icon, that the icon always shows "speaker disabled" at power up, yet speaker audio is heard if an app such as "Audio" is started.  If subsequently the top-line speaker option is toggled On then Off while the app is running, the speaker becomes silent.  However, if the app is exited and restarted, audio output is again heard even though the top line would seem to indicate that the speaker is disabled.

Perhaps should the H1 Speaker on/off setting be preserved across power-cycles in pmem?

If audio output is disabled per this setting, shouldn't the Audio app check this setting (vs turning on the speaker output anyways whenever the app is executed)?

sizzlik, 2023-06-06:
> I'm not sure of the expected behavior, but on my H2+ I observe (using latest nightly) that if the H1 Speaker Option is set to "show" the icon, that the icon always shows "speaker disabled" at power [...]

Exactly what i was wondering. Audio plays even thou its set to silent on startup. Turning the sound on and off again its muted..until next time. I think the audio should be turned off at all time, unless you want to listen in and then press the according button to do so.

NotherNgineer, 2023-06-06:
@kallanreed what do you think about this?  (I'm sure you're very familiar with this code after adding the persistent audio volume feature.  :-)  Should the "set_speaker_mode()" function store the mute/unmute setting in pmem?

kallanreed, 2023-06-06:
I was just messing around with this this morning. Some apps explicitly mute/unmute and I think that’s a bug. There’s also some difference between speaker audio and headphone audio in the API but all the apps only use the headphones setting.

IMO only the settings should be changing the mute state (much like the bias_t setting), but there’s one weird case where switching some modes (like loading a img or changing the modulation) where a mute may be needed. I’m those cases the app will unmute afterwards which ignores/changes the setting.

NotherNgineer, 2023-06-07:
@kallanreed  Some apps support "squelch" which needs to mute audio temporarily using another mechanism when signal level is below the squelch setting.  BTW, I should have mentioned squelch before because that's one more audio setting that I would love to see stored in persistent memory so I wouldn't need to fiddle with it every time I run the Scanner app.

And, of course, audio needs to be muted temporarily when receiver is disabled, yet never turned back on if the "H1" navigation-bar speaker setting is set to mute.  I guess I see the navigation-bar speaker speaker setting as a means of disabling audio & saving power on boards with no speaker, which should be stored in persistent memory, and should not be overridden by any app.

As I only have an H2+, I am just now looking at these two Wiki pages below regarding speaker differences between H1 & H2 to understand how that navigation-bar speaker was originally described to work.  Now I'm left wondering if power could be saved on my H2+ if the CS8122S amplifier chip could be shut off somehow when using headphones, and/or if the AK4951 chip could be shut off when not using headphones.  If code is only messing with the AK4951, does that imply that H2 audio goes through the AK4951 chip to get to the CS8122S?...  Lots of questions...  :-)

  https://github.com/eried/portapack-mayhem/wiki/Internal-speaker
  https://github.com/eried/portapack-mayhem/wiki/Differences-Between-H1-and-H2-models

NotherNgineer, 2023-06-07:
@ZxSpectruM55  Thank you for pointing that out.  I have an H2+ board and I see the same thing, both with 1.7.1 firmware and the latest nightly.  It's my understanding that there are two separate amps, one for the headphone and one for the speaker, so I "hope" it's possible for software to turn off the speaker when headphone is in use.  We'll need to do some experiments...

NotherNgineer, 2023-06-14:
Some things I found out from disassembling my H2+:

A WM8731 audio CODEC is installed in place of the AK4951. The WM8731 IC has no speaker amplifier, for one difference.

An LTK8002D audio amplifier is installed in place of the CS8122S.  Either of these speaker amplifier chips have a "Shutdown" input but not sure what it's connected to.  The "Line Out" signals from the WM8731 are presumably attached to the LTK8002D audio inputs (it's hard to get to the LTK8002D under the corner of the glued-on battery).

Inserting a headphone plug does not electrically disable the speaker output, so it's left up to (future) firmware to do this.  The headphone jack is a 4-conductor plug for microphone input in addition to right/left headphone audio outputs.

NotherNgineer, 2023-06-16:
I'm attaching a binary that supports persistent audio muting via the icon on the navigation bar.  I've tested it on my H2+R4 with the WM8731 codec, but it would be very helpful if folks with an AK4951 codec could give it a try and let me know.  Note that whenever "mute" is enabled that both the speakers and the headphones should be muted.

[portapack-h1_h2-mayhem.bin.zip](https://github.com/eried/portapack-mayhem/files/11774174/portapack-h1_h2-mayhem.bin.zip)

NotherNgineer, 2023-06-17:
I'm attaching a second binary for testing below.  This just has some code simplification, but should function the same as the first test version above.  If anyone has a chance to try these out, please let me know the results.  Thank you!

[portapack-h1_h2-mayhem.bin.zip](https://github.com/eried/portapack-mayhem/files/11777905/portapack-h1_h2-mayhem.bin.zip)

Brumi-2021, 2023-06-17:
Hi @NotherNgineer thanks ,
I will check this 2nd version later this afternoon in both platforms AK / WM and report to you . 
Cheers

ZxSpectruM55, 2023-06-17:
@NotherNgineer 
I think this bin file is a test solution for the first problem. There is no problem on my h2+r4 device. When I plug in the headphones, the sound continues to come from the speaker. I think we can't find a solution to this problem.

NotherNgineer, 2023-06-17:
@ZxSpectruM55
Thanks for confirming the fix for the original issue above.  With H2+R4, I think a hardware fix might be needed to disable the speaker when the headphone is plugged in (the wiki has instructions here but I don't want to risk breaking my board: https://github.com/eried/portapack-mayhem/wiki/H2-Plus-speaker-modifications )   I did email the seller that I bought the board from, looking for a schematic, but no reply.  :-)

NotherNgineer, 2023-06-17:
I've added one more audio feature in the attached bootleg (and in PR #1161), and that is the ability to disable the AK4951 speaker amplifier in the Settings -> Audio menu.  Disabling the speaker amp saves a few mA on AK4951-based H1R2 or H2R1 boards in cases where no speaker is installed or only headphones are used.  This setting has no effect on board that have the WM8731 codec.

[portapack-h1_h2-mayhem.bin.zip](https://github.com/eried/portapack-mayhem/files/11779782/portapack-h1_h2-mayhem.bin.zip)

Brumi-2021, 2023-06-18:
Hi  @sizzlik ,
After studying it , unfortunately we can not mute the spk alone, keeping the HP in the WM audio codec.
But this could be applied in AK audio codec. 

And in fact , as you could see, thanks to @NotherNgineer , this feature for AK has been tested and merged,then  I think that now,  we can close that issue . 

And  regarding your other question, : „What does the H1 Key tone option do thats set to 20%? „ 
 This default 20% is the modulation % of the subcarrier CTSCC pilot tone that we can add in the Mic App when transmitting in FM . 

If you agree, @sizzlik ,  pls could you close that issue ? 
thanks

NotherNgineer, 2023-06-19:
Just wanted to clarify one thing (since a lot of this was just discussed in Discord):

The way it's "supposed" to work after PR #1161 is as follows:

  Mode 1: "Disable AK speaker" is unchecked, or WM-based board:  MUTE icon controls both speaker & headphones.

  Mode 2: "Disable AK speaker" is checked AND it's an AK board:  Speaker is disabled, and MUTE icon only controls headphones.

I am thinking of making some additional revisions, but this is how it's supposed to work as of PR #1161.

sizzlik, 2023-06-19:
My issue wasnt about headphones, more about that the speaker is in general not muted on startup, even if the H1 audio button is grey (muted) on startup. To mute it you got to enable and disable each time you start an app. Its no big deal if it cant be changed. Thank all you guys for looking into it and putting in efford to fix it. -Issue Closed-

closed 2023-06-19 by sizzlik