portapack-mayhem/mayhem-firmware #1101: Whos Bug is it? Reset button freeze

sizzlik, 2023-06-03:
Hi,
I got a pre-assembled device from OpenSourceSDRLab on Ali. (hRF r9 + PP H2+) It came with the usual "Thanks 4 Purchase" note plus some additional information/warning:
- "Must use Nightly builds .. Stable releases dont work atm." That problem seems to have been fixed as i installed the latest stable without problem. (But back to newest nightly now)
- "Pressing the reset button freezes the device. Wait for firmware fix" That one still does exist and the guys wrote on the note it would be a firmware issue they cannot fix and i have to wait for you guys to fix it.

Now my question..is that something that can be fixed in firmware or do they try to blame others for issues they caused?
As its battery powered, the situation currently is: One encoder click to turn on, double encoder click to turn off, encoder click does not work to select menus/options (gotta use the tac button..bit sad) and pressing the reset button freezes the device. DFU button works.

eugene210492, 2023-06-04:
same problem , portapack h2 "PCB v3.6 mdv.club" ( the AG256SL100 IC)

Brumi-2021, 2023-06-04:
Hello @sizzlik , I totally agree , I think it is a problem of H2+ board with integrated  battery , and how they handles from Portapack  the HackRF CPU power and it’s micro reset .

in my case , H2+ Portapack  and big TFQFP 100 CPLD it produces picture freeze and totally hang up. (Only possible to recover by switch off-on by a short push to knob encoder button) .

In other H2+ Portapack with original TQFP64 CPLD, based on @Sommermorgentraum test,   it produces black picture and also hang up .

BTW, recently , thanks to oa-1, he  shared a paper instruction manual that came to his H2+ , and they clearly recognised  that RESET freeze problem , but they blame to Mayhem , that i totally disagree. 

[image]

I think we can not fix that problem from Mayhem  side. Moreover , to to it at least we will need to see H2+ schematic , and as far as I know , they never shared it.

Then, at the moment we have to,leave with that problem , and just use the short push to the know encoder , to make power on and power off . And actually to go to DFU mode , you will need to switch off the device , and ,keep,pressing DFU button while you switch on with a short push to the know encoder button . 

If you do not have more questions , I would suggest to close that well H2+ PCB 3.6  known problem. 

Cheers

sizzlik, 2023-06-04:
yep..thats similar to the note i got. Already thought its not a firmware problem. No further questions, can be closed.

closed 2023-06-05 by gullradriel