portapack-mayhem/mayhem-firmware #1103: Large file read/write block size corrupts data.

kallanreed, 2023-06-03:
There's a comment in PNGWriter.cpp indicating working around a file write problem.

```
    // Small writes to avoid some sort of large-transfer plus block
    // boundary FatFs or SDC driver bug?
```

While writing the screenshot viewer, I also ran into this problem if I tried to read a block of 720 bytes all at once. The data was _mostly_ correct, but there was clearly corruption in the data (visible when rendered). I also worked around it by reading in chunks of 240 bytes.

I have not investigated too deeply, but there's likely a bug in FatFs (less likely), or the SD Card driver, or perhaps there's a documented size limit that we need to respect.

gullradriel, 2023-06-04:
A limit sounds like a valid explanation, you go over and read garbage

gullradriel, 2023-07-18:
I think according to https://github.com/eried/portapack-mayhem/blob/92072b4225f4f2fae95a60e40fc8b21a36352ebe/firmware/common/ffconf.h#L158 its not a bug.

Documentations about these variables and a quick look at them in the code show that the firmware is configured for a fixed 512 block size . It may work with lower read/write sizes, but no with anything bigger than it.

https://github.com/eried/portapack-mayhem/blob/92072b4225f4f2fae95a60e40fc8b21a36352ebe/firmware/chibios-portapack/ext/fatfs/src/ff.h#L170 indicate that _MAX_SS is also used to configure an internal buffer used in read/write operations.

I think we can close this, but we have to make a Wiki entry about it, in dev section.

gullradriel, 2023-07-18:
And yes, from the look of it 512 seems to be the max block size.

gullradriel, 2024-01-05:
Done added the doc and referenced the issue.

closed 2024-01-05 by gullradriel