portapack-mayhem/mayhem-firmware #1111: Batch Replay TX

M00N66, 2023-06-05:
Imagine being able to make a playlist of replay attacks. For instance:
Ceiling Fan light, garage door light, stereo skip, etc. And then the loop function could keep cycling the files, Maybe even add a randomize order feature. Would be great for haunted house pranks and film effects.

Brumi-2021, 2023-06-08:
Hi @M00N66  ,
 as @zxkmm mentioned you 
Did you check or evaluate the existing Playlist App inside TX App’s ?
Is it matching your expectations? 
If so, can we close that issue ?

Thanks

Brumi-2021, 2023-07-02:
Hi @M00N66 , 
I  think the new  Replay is already a fusion of previous Replay + Playlist able to do what you were requesting for.  
Pls check the wiki, and try it  , and let's close that issue. 
If you miss any other function, feel free to reopen a new issue . 
Cheese

closed 2023-07-02 by Brumi-2021