portapack-mayhem/mayhem-firmware #1113: Latest FREQMAN app anomalies

NotherNgineer, 2023-06-06:
Using the latest nightly, I've noticed the following anomalies in the FREQMAN app:

1) The "New" option limits category (file) name to 10 characters, which seems a tad short (vs existing Freqman files that come on the SD card).

2) If the "New" option is used to create a new category (file), FREQMAN oddly "jumps" to a different category (file) immediately after file creation, and the newly created file cannot be found in the list until the FREQMAN app is exited and restarted.

3) After a new empty FREQMAN file is created, FREQMAN doesn't seem to provide any way to manually add a new Freq/Description to said file ("Edit" options are missing), or add a new frequency line to existing files for that matter.  The new "Notepad" app could be used instead, but I would have expected this ability in FREQMAN.

4) Entries in the frequency list are not being painted correctly.  When stepping through the list, the currently selected frequency is not highlighted (this worked correctly in v1.7.1).

5) After an existing FREQMAN file is opened which does contain a list of frequencies, the "Edit" options for Frequency and Description don't seem to be working.  After I change either, no change appeared in the list or in the file, whether I pressed "Save" or "Done" afterwards.  (I might perhaps expect an inability to edit a file that's too large, but I tried this with a small file)  [On subsequent attempts, even using v1.7.1, I got a "M0 Guru meditation out of memory" error which could have something to do with this TBD.]

6) "Save" and "Done" are a bit ambiguous when editing a frequency value.  Does "Done" mean Cancel, or is it the same as Save?

gullradriel, 2023-06-06:
First please fix the title of your issue. I don't think most of your issues appeared with the nighlty, are they ?

gullradriel, 2023-06-06:
For all the other issues, your welcome to push a PR :-p

NotherNgineer, 2023-06-06:
Please accept my apologies if the title or comments seemed aggressive.  During my brief testing, I was surprised by some of the behaviors observed.  I didn't mean to point blame at anyone.  Perhaps there is simply room for some enhancements, if code space permits.

Referring to numbers above.
1)  OK.  When entering the file name it seemed that there was plenty of room for more characters, but not so much room on the main screen between "Category:" and the "New" button.
2)  Same behavior in 1.7.1.
4)  Personally I don't like the style change.  It was much easier to see which line was selected using v1.7.1.

gullradriel, 2023-06-07:
Hello there,
-Highlight is back
-Edit freq & descriptions and button delete are working
-When editing/deleting, position inside the loaded category is kept
-When creating a new category, the category list is reloaded, just scroll to see your new category
As for the filename and description lengths I hope you understand that I will not change them to preserve a bit of memory
Please test in next nightly or update your next branch.
I'm closing that issue, next time if you want to open a new issue on FreqMan, you will be able to call your issue 'new freqman problems/enhancements' as that time I killed/rewrited a good bunch of it ;-)

gullradriel, 2023-06-07:
Will be finally fixed correctly when https://github.com/eried/portapack-mayhem/pull/1120 will be merged

gullradriel, 2023-06-07:
Got most of them :-)

NotherNgineer, 2023-06-07:
@gullradriel   I've coded a fix for the above and will submit a PR in a few.

gullradriel, 2023-06-08:
I've merged your proposition, thanks :-)