portapack-mayhem/mayhem-firmware #1147: Pocsag Receive gives error

The-Font, 2023-06-12:
Since version 1.7.2 I can not use pocsag receive anymore. 

I can not navigate to change the frequency. If I press the middle button then I get a M0 Guru Meditation error.

The-Font, 2023-06-12:
Hi @Brumi-2021 
Thank you for the tip. Pressing 3 times up does help and I can goto the freuency and change it. I also receive messages then.

The-Font, 2023-06-12:
Hi @loewal, I will try this build and let you know the results. I am also already happy with the workaround that was posted here.

loewal, 2023-06-12:
I used that too 😀👍

closed 2023-06-12 by The-Font

Brumi-2021, 2023-06-12:
Hi, glad to see that you or we helped to  fix it and closed it . 
Just for our reference , which is a popular default freq. to be used in Pocsag  ? 
our version N_230612 has already a good default one ?  (In my country I can not check it, because it was discontinued many years ago) . Thanks 

Cheers

loewal, 2023-06-12:
I my countrY (The Netherlands) it’s: 172.450 as you can also check in the Frequency Manager.

See ya

pickeditmate, 2023-06-17:
FWIW 153.350 is popular here in the UK.

I also get hits on 
137.9775
153.4250
454.0750
454.200
459.0440