portapack-mayhem/mayhem-firmware #1150: TPMS app works only once

DavidIgor, 2023-06-13:
---- latest mayhem_nightly_n_230613_FIRMWARE. is making tpms app work only once , as you boot device up and open tpms app all works , however if you close the tpms app and reopen it it wont detect signals . 

(Please try the latest nightly release before submitting this. You can find the latest nightly version here: https://github.com/eried/portapack-mayhem/releases)

----

**Describe the bug**
latest mayhem_nightly_n_230613_FIRMWARE. is making tpms app work only once , as you boot device up and open tpms app all works , however if you close the tpms app and reopen it it wont detect signals . 

**To Reproduce**
Steps to reproduce the behavior:
1. turn on portapack
2. open tpms app
3. test it with car or signal 
4. go back to man menu of portapack
5. reopen tpms app
6. test with car or signal
7. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Affected versions**
mayhem_nightly_n_230613_FIRMWARE

Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release:
* Latest Nightly release:
* Previous working release:

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.

gullradriel, 2023-06-13:
https://github.com/eried/portapack-mayhem/pull/1151 may solve this

NotherNgineer, 2023-06-13:
@DavidIgor
Can you tell me if it's the 315MHz or 434MHz sensors that you're seeing an issue with?
Also, under Settings->AppSettings, do you have either of the Load or Save boxes checked? (it should work either way but I was just trying to reproduce the issue)

DavidIgor, 2023-06-14:
@NotherNgineer hi, none of  app settings file is checked, also my sensors are 433MHz

NotherNgineer, 2023-06-14:
I just fixed a recently-introduced issue in PR #1152 where the receiver frequency wasn't being programmed when initializing the TPMS app if App Settings are disabled.  I thought this would only affect 315MHz sensors though, since toggling the frequency between 315/433 should have caused the receiver frequency to be tuned correctly.

closed 2023-06-14 by gullradriel

NotherNgineer, 2023-06-14:
@DavidIgor 
Please try the attached firmware (or the NEXT upcoming nightly build n_230615) and let us know if it resolves the issue.  (If it doesn't, then please re-open this issue.)

[portapack-h1_h2-mayhem.bin.zip](https://github.com/eried/portapack-mayhem/files/11748499/portapack-h1_h2-mayhem.bin.zip)

DavidIgor, 2023-08-17:
@NotherNgineer Hi , latest release and also 1.7.4 does not read renault TPMS in tpms app. 
it can pick up other tpms sensors tho. 

Something must have changed as I was able to read my tpms sensor without issues(and still can with flipper zero) .

NotherNgineer, 2023-08-17:
While attempting to reproduce your issue, I just tested the TPMS app on both of our vehicles, one on 315MHz and one on 433MHz (need to set the frequency manually in the app), with and without "App Settings" enabled, and sensors in both vehicles were reported OK.

Unfortunately the TPMS app does not support all sensor models, and I don't have a Renault to test it with.  If there is an older version of Mayhem firmware that does support the Renault TPMS sensors, please try to determine exactly which Nightly release it stopped working in.  Also it might help to compare the LOGS\TPMS.TXT log file between a version of firmware that works with a version of firmware that doesn't.

Does the app work better with "App Settings" disabled?  If so, you might try deleting the "SETTINGS\rx_tpms.ini" file so the settings will get reinitialized.

Was the TPMS app the first app executed after power-up?  (Shouldn't matter of course, but there could be a bug like before.)

If none of that helps, does Flipper Zero have any sort of logging capability that might contain useful info?  Some additional TPMS sensor types are supported in the linux rtl_433 app (more than in Mayhem), and it does support verbose logging which might contain more info that could help to debug this.

DavidIgor, 2023-08-17:
@NotherNgineer frequency set to 433 as Im in EU . 
no matter if tpms app runs first after boot up.
deleted setting file for tpms on sd card but no change. 

I will downgrade and report back

nase111, 2024-09-23:
Hello. There is no frequency 433000000 on the TPMS_RXapp, but it is set to 433920000, which is probably the main problem. Even if I overwrite this frequency in settings/rx_tpms.ini to 43300000, it will not help, because the application will overwrite it again to 433920000.

NotherNgineer, 2024-09-23:
I am unaware of any TPMS sensors that transmit on 433.0 MHz (as far as I know, "433" in a product description means 433.92 MHz, and theoretically the 1.75 MHz bandwidth should pick up the sensor either way).  In the USA, sensors have an FCC ID printed on them and one can look up the specs in the FCC database https://apps.fcc.gov/oetcf/eas/reports/GenericSearch.cfm  

To test the theory that maybe a sensor is transmitting on 433.0, you could go into Settings -> Freq Correct and temporarily set the RX correction value to -0.92 MHz to subtract out that 0.92.

I think it's more likely a different protocol.  But if you find any TPMS data sheets that say 433.0 MHz, then we can certainly add that frequency to the list.

nase111, 2024-09-24:
OK, thank you very much, I will try it.

nase111, 2024-09-24:
I tried, it didn't show the data. My car is a Suzuki Vitara 1.4 Boosterjet, year of manufacture 2018, assembled in Hungary, i.e. EU version. It will probably be an unknown protocol for you.

htotoo, 2024-09-24:
Any older fw version worked with it? 
if not, then it may be an unknown protocol.

nase111, 2024-09-24:
I haven't tried with the older version, but I don't believe it would work.

PR #1153 (2023-06-14): Starting freq fixes when no App Settings, and support for App Settings in Scanner App (Mode/BW/Step only)
This is a revised version of closed PR #1152 after I've merged my changes with PR #1151.  May fix #1150.