portapack-mayhem/mayhem-firmware #1154: The system is down

utenus, 2023-06-14:
**Describe the issue**
After updating to version 1.7.1 according to the instructions from GitHub, HackRF does not start at all. When I try to turn it on, the LEDs light up, but PortaPack does not show anything on the screen.

**My Hardware**
Please specify what PortaPack hardware version you are using. 
[Hardware](https://de.aliexpress.com/item/1005004737929723.html?aff_fcid=8d0f6bef00674426b50da7b304d90015-1686757182242-08189-_DmU7GQX&tt=CPS_NORMAL&aff_fsk=_DmU7GQX&aff_platform=shareComponent-detail&sk=_DmU7GQX&aff_trace_key=8d0f6bef00674426b50da7b304d90015-1686757182242-08189-_DmU7GQX&terminal_id=eeb88b1c7c6249e9bf72f5b070e8279e&afSmartRedirect=y&gatewayAdapt=glo2deu)

**Affected versions**
1.7.1

**Were you able to update the firmware before?**
Previously, the update to version 1.7.0 went smoothly.

NotherNgineer, 2023-06-14:
You might want to try this, but I don't know if it'll help:  https://github.com/eried/portapack-mayhem/wiki/Won't-boot

pickeditmate, 2023-06-17:
Did that solve your problem @utenus ? Can this be issue be closed now?

gullradriel, 2023-07-18:
No answer for too long.

closed 2023-07-18 by gullradriel