portapack-mayhem/mayhem-firmware #1176: Additional Map Zoom enhancement ideas

NotherNgineer, 2023-06-21:
Some potential map zoom improvements for phase 2 (I'll be looking into these):

1)  Add in bottom right map , the applied zoom , x1 ,x2, x4, ....

2)  When user decrease lat or long minutes and reaching 0 , it could reduce 1 degree and continue with 59 minutes. (or when increase and reaching 59 , it could add 1 degree and continue with 00 minutes.

3)  As plane signal goes away, fade it away on the map slowly perhaps, like in the list.  But, should we guess the new position of the ghost plane based on time and previous speed & heading???

4)  Show a "trail" of dots, or a line, behind each plane indicating where it has been while the map has been displayed (showing the flight pattern).

5)  Consider whether the zoomed map should be reprocessed to smooth out the transitions between the new larger pixels (not sure if we have processor power...).

(Thanks to @Brumi-2021 for suggesting several of these)

htotoo, 2024-01-01:
Yeah, I miss the Zooming out function, since the adsb reception is too good, it doesn't fit the screen :)

NotherNgineer, 2024-01-04:
I've added simple Zoom Out support in PR #1710 for you to try out...

htotoo, 2024-01-05:
Works like a charm!
Also i think we don't need to avg pixels, since with this small display it is good as it is now. Also without it we save a lot of cpu power, and can zoom out further and faster.

NotherNgineer, 2024-01-16:
PR #1767 adds support for #2 above:  When user decrease lat or long minutes and reaching 0 , it could reduce 1 degree and continue with 59 minutes. (or when increase and reaching 59 , it could add 1 degree and continue with 00 minutes.

closed 2024-01-16 by gullradriel