portapack-mayhem/mayhem-firmware #1186: Change color of Yellow icons  (please assign to me)

NotherNgineer, 2023-06-24:
@Brumi-2021 suggested that we change the color of the Yellow Icons so that the icon is still visible when selected.  Perhaps Orange would be a better choice than this.

[image]

Brumi-2021, 2023-06-24:
Hello thanks for your reply. 
Yes I think so , orange icons looks like very visible (confirmed in both screens H1, H2), much more than yellow.

it is minor improvement, but I think all members will agree , pls  proceed with it, 
Cheers

zxkmm, 2023-06-25:
thx for mentioning. that was originally decided by @sck-nogas , however I think it doesn't matter anymore, so it's all good for new color scheme, even if you decided to use rainbow color XD. and the color doesn't bog me at all. 
ref. https://github.com/eried/portapack-mayhem/wiki/Applications

jLynx, 2023-06-25:
So the colour was the state of the application, for example if it's just a poc, almost complete or done (green)

jLynx, 2023-06-25:
https://github.com/eried/portapack-mayhem/wiki/Applications

NotherNgineer, 2023-06-25:
Resolved in #1189

closed 2023-06-25 by NotherNgineer