portapack-mayhem/mayhem-firmware #1197: Firmware version disappears from home screen (very low priority)

NotherNgineer, 2023-06-27:
This is just a minor peculiarity, and has existed since at least FW version 1.7.0.  It should be very simple to fix, but writing it up anyway to remind me to look into it.

Steps to reproduce:
1)  Turn on Portapack
2)  Press UP, then RIGHT buttons to move focus to one of the first 3 icons on the screen.
3)  Press DOWN button, and the firmware version disappears from the bottom of the home screen and is replaced by a black box (see screenshot below).  (The time also disappears briefly but reappears in a second when the time is updated)

Expected behavior:  FW version remains visible on home screen, or is re-painted afterwards if the screen is cleared.

[image]

Brumi-2021, 2023-06-27:
Hello I could reproduce it , doing the following  (slightly different as you said in point 3  ) , 

1. same as you described. 
2. same as you described.  
3. From <Transmit App icon >  Press UP, then 3 x LEFT buttons to move focus to the first left  icons on the title bar (camera icon) .
Press DOWN button, and the firmware version disappears from the bottom of the home screen and is replaced by a black box (see screenshot below). (The time also disappears briefly but reappears in a second when the time is updated)
4. same as you described. 

Cheers,

NotherNgineer, 2023-07-03:
Additional findings:

1)   It looks like the whole screen is being re-painted (except the FW version) when moving focus down from one of the left-most icons to the left app (RECEIVE) "button", and the whole screen is not re-painted when moving focus down from the other icons to the right app (TRANSMIT) button.

2)  It's not just the main menu screen.  The re-painting occurs on EVERY menu screen whenever focus is moved from left-most icons on the the title/bar to the upper-left app button.

eried, 2023-07-03:
Strange also that is actually replaced by an actual black box instead of not being displayed. Thanks for the report

sizzlik, 2023-07-05:
Replace the black bar with "You found an easteregg"

NotherNgineer, 2024-04-07:
Problem still exists

commit a21e9ca1 (2024-09-26): Nav fix, fixes #1197 (#2270)

* Nav fix, fixes #1197
closed 2024-09-26 by htotoo