portapack-mayhem/mayhem-firmware #1198: ERT RX app seems to be broken (n_230618 and newer)

NotherNgineer, 2023-06-27:
It appears that the ERT RX app has receive issues in firmware versions n_230618 through the latest n_230627.
Some signals are received in the ISM band and stored in the log, but the bit error rate is extremely high with the latest firmware, resulting in no ERT meters being shown in the app.

To test, I have tried both enabling & disabling the "Load App Settings" & "Restore App Settings".

I'll investigate further tomorrow since it's late here.

NotherNgineer, 2023-06-28:
This issue was introduced by the compiler optimization level change in PR #1182.

firmware/baseband/CMakeLists.txt

I have tested that the issue goes away if I change the optimization level back to -02, but unfortunately we're now out of ROM space unless we jettison some stuff, or only use the -O2 for the ERT baseband module.

gullradriel, 2023-06-28:
Should be fixed Ith https://github.com/eried/portapack-mayhem/pull/1206 f

closed 2023-06-28 by gullradriel