portapack-mayhem/mayhem-firmware #1205: Unistall app

samukas81, 2023-06-28:
Add option to uninstall app.  It has a lot of useless app.

NotherNgineer, 2023-06-28:
Would you like to be able to rearrange the apps also, like a sliding tile puzzle, to put the most frequently used ones at the top?  (Maybe we could use my "long press" support to select an app to be moved)

samukas81, 2023-06-28:
Hello I would like to uninstall some program is it possible?

NotherNgineer, 2023-06-28:
Currently the only method to disable apps is as follows:

1)  Download the source code and compiler as described on this wiki page:  
https://github.com/eried/portapack-mayhem/wiki/Compile-firmware

2)  Edit the file firmware/application/ui_navigation.cpp, look for any line that has "nav.push<" on it, and you can add "//" to the beginning of the line to disable the app mentioned on that line.

3)  Rebuild and flash your custom firmware image using the instructions on the wiki.  (After compiling, the new firmware image portapack-mayhem-h1_h2.bin can be found in the folder build/firmware)

I know you'd prefer to see something in the Settings menu to do that, but this might be a way to do it in the interim.

samukas81, 2023-06-28:
Ok, as I don't understand programming I'll wait for someone to create this uninstall tool, but thanks for the feedback.

jLynx, 2023-06-28:
We will add this in the future when we work out how to save the apps to the SD card rather then have them built into the firmware as we will need to do this since there is no space left on the device

NotherNgineer, 2023-07-01:
> We will add this in the future when we work out how to save the apps to the SD card rather then have them built into the firmware as we will need to do this since there is no space left on the device

@jLynx  You'd think I'd know this already, but is the M0 code running from ROM now or is it copied into RAM at power-up and executing from there?  I was just considering that we're low on RAM space too, so when we load apps from the SD card as needed, are we going to swap them out in Flash ROM or load them into RAM?

jLynx, 2023-07-04:
@NotherNgineer possibly a solution. I honestly haven't looked into it too far yet as I have been busy with the CPLD stuff

PR #1502 (2023-10-15): Hide unwanted apps by naming them in SETTINGS/blacklist file
Simple support for reading and checking a SETTINGS/blacklist file before showing app icons.  Apps can be hidden by listing them in the file using their exact name as it appears on the menu screen.  Changes take effect on the NEXT boot.

This may fully resolve #1205, and partially help with #1492.

The blacklist file must be edited manually -- there is no fancy settings screen.

WARNING:  Don't get carried away and create some massive blacklist file or you'll run out of memory during power-up and will then need to delete the file off  [... cut]
closed 2023-10-15 by NotherNgineer