portapack-mayhem/mayhem-firmware #1214: "SD Over USB" issue after compiler optimization changes in n_230629

NotherNgineer, 2023-06-29:
The compiler optimization changes in PR #1206 had the unintended side effect of breaking "SD over USB" starting in the n_230629 build.

This is a KNOWN issue.  Just documenting for tracking purposes, and a place (besides discord) to document information learned about the problem.

Perhaps one possible solution is to revert the optimization change for the "SD over USB" code only.

eried, 2023-06-30:
But in the long run, do we revert the optimization change or not? 🤔

gullradriel, 2023-06-30:
We reverted optimisation for all until we find a way to make it work for gcc 12 and with -flto.
Actual next and incomming stable are back with no -flto and -O3

NotherNgineer, 2023-07-01:
Closing since optimization changes were reverted -- for now

closed 2023-07-01 by NotherNgineer