portapack-mayhem/mayhem-firmware #1228: Display CTCSS tone frequency in Audio RX app

NotherNgineer, 2023-07-01:
It has been suggested to display the CTCSS tone frequency (in addition to the existing CTCSS code numbers) in the Audio app.

This enhancement has been discussed in Discord; just submitting this ticket for a reminder/tracking purposes.

@Brumi-2021

Brumi-2021, 2023-07-01:
Thanks @NotherNgineer ,
 Let me summarised what we discussed + some personal opinions , but feel free to add any other opinion

I-) if possible , it would be preferable to show both information at the same time (CTCSS code #,  and Measured tone frequency. Hz ) than alternative. (Therefore we will need to make display  space , squeezing the SQ char space or redistributing the Rec line , SD free recording time , file name .)

ii-) it would be appreciate if we can reduce the flickering freq decimal digits. (Reducing refresh display time or avg ) is 

iii-) it will be nice to display the mayhem calculated freq, even it has some small absolute error (<0,4%)  compared to the real transmitted one . (Because it will useful to debug with any other walkie-talkie incompatibility.

iv-)here another open discussion about how to handle : when there is not any transmission , the noise is producing a continue flickering data and “????” , maybe it would be better just mute or blank that label in those conditions ? 

V-) CTCSS PMR or repeater tones are placed  under sub- voice spectrum < 300 Hz , so we will not decode or measure any other subcarrier tones like 19khz, 32k …that Mic App has . 

Feel free to add any other comments or relevant opinions , related to that new feature spec ,

samukas81, 2023-07-01:
I believe that displaying only the frequency is enough, the default selection of radios is frequency.

NotherNgineer, 2023-07-01:
> I believe that displaying only the frequency is enough, the default selection of radios is frequency.

Thank you for saying that.  I agree with you.  When there's limited space I'd prefer to see the tone frequency vs a non-standard code number and the Motorola PL code.

NotherNgineer, 2023-07-03:
Enhancement completed in PR #1231

closed 2023-07-03 by NotherNgineer