portapack-mayhem/mayhem-firmware #1229: Screen Mirroring with control over USB

hackwin, 2023-07-01:
**Is your feature request related to a problem? Please describe.**
Control Portapack remotely from a PC

**Describe the solution you'd like**
Stream display and control data over USB so the Portapack GUI can be controlled from a PC
Similar to VNC or MS Remote Desktop
The Flipper Zero does this in their qFlipper ProtoBuffer https://github.com/flipperdevices/flipperzero-protobuf
Arduboy also does it.  I wrote a small demo for it here: https://community.arduboy.com/t/screen-mirroring-guide-how-to-screenshot-your-game/2800/18

**Describe alternatives you've considered**
SDRSharp, GNURadio, and various other programs can work similar to Portapack.
However, I want to control the Portapack without using the physical controls

**Additional**

sizzlik, 2023-07-05:
Had a similar thought. Would be a lovely feature, but i guess you got to code some sort of companion app that does the job. Like turning the firmware into software. Would be cool to use the portapack firmware apps with a hackrf that got no portapack and is just in an enclosure. Not quite the scope of the project, but with the geniuses at work here we can hope a fork pops up =)

hackwin, 2023-07-07:
An app would need to handle receiving screenshots and simulating button presses.  Screenshot functionality exists already.  Maybe a Java app for any OS would work.

htotoo, 2023-12-01:
I can imagine this via usb serial mode (high baud). also file explorer commands would be great for this app.

bernd-herzog, 2023-12-17:
fixed in https://github.com/eried/portapack-mayhem/pull/1648

closed 2023-12-17 by eried