portapack-mayhem/mayhem-firmware #1232: Squelch inconsistencies and persistency

NotherNgineer, 2023-07-02:
It would be good if Squelch could be made more consistent across apps, and be supported in all modulation modes, and if it could be saved persistently.

Currently:
1)  In some apps, Squelch is a positive number, and in other apps it's a negative number.
2)  Squelch is not saved in the App Settings (for at least some apps) or in persistent memory.
3)  Receiver and most apps only support Squelch in NFM Modulation (notably Scanner supports Squelch in all modulation modes but doesn't use a Receiver setting to do it).

NotherNgineer, 2023-08-23:
Update:
Item 2 & 5 above can be considered completed:  PR #1394 saves Squelch in settings file for Scanner & Recon.  For other apps the NFM squelch has already been saved in App Settings.

eried, 2023-12-05:
I am closing issues without much discussion to keep this clean, but feel free to reopen this again if needed.

closed 2023-12-05 by eried