portapack-mayhem/mayhem-firmware #1233: AIS RX not working after ADS-B RX

yellowz1234, 2023-07-02:
**Describe the bug**
AIS RX does not work when it is launched immediately after an ADS-B RX session. It does not crash, but it is just not showing any vessels.
When AIS RX is launched immediately after turning on the PortaPack or after an Audio RX session, there is no problem and it shows vessels as expected.

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on PortaPack with Mayhem 1.7.3 (on HackRF One R9 and PortaPack H2+ R4).
2. Go to AIS RX: everything works fine and I can see multiple vessels.
3. Close AIS RX without turning off PortaPack and open ADS-B RX: everything works fine and I can see multiple airplanes.
4. Close ADS-B RX without turning off PortaPack and open AIS RX again: no vessels are detected.

**Expected behavior**
After closing ADS-B RX and starting AIS RX, I'm expecting to start receiving boats as normal, without having to turn off and on the PortaPack.
However, without restarting the PortaPack, AIS RX is no longer detecting anything, no matter how long I wait, until the PortaPack is restarted.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release: bug first observed with 1.7.3
* Latest Nightly release: Nightly release - 2023-07-02: same behavior as 1.7.3
* Previous working release: not tested

**Additional**
Is it possible that the AIS RX frequency is not set correctly after running ADS-B RX?

Brumi-2021, 2023-07-02:
Hi @yellowz1234 i can  NOT  reproduce your problem .
I used also 1.7.3 and latest nightly and both works well .

AIS Rx is working always well even using before ADS-B or not . I do not see any issue . No difference I. Terms of AIS capture from direct booting or after used ADSB .

More over , now we can check pressing two times DFU button the Rx tunned freq in each Rx App , and always is ok . 

Can you recheck and give us more details ? 

Thanks

NotherNgineer, 2023-07-03:
From pressing the DFU button twice and looking at Receiver settings, it looks like there's a bug to me:

There are no boats around me to test this, but I turned off "Load App Settings" and "Save App Settings" (under Settings->App Settings) and it looks like the modulation type is 1 (NFM) when the AIS app is the first app run.  But if I run the ADS-B app first, it sets the modulation type to 3 (SPEC), and that gets carried over to the AIS app.

I also notice that if I have enable "SAVE App Settings" only that the modulation type is saved as 1 in the file if the AIS app is the first app run after power-up.  Then if I subsequently enable "LOAD App Settings" also, it will always load a 1 into the modulation type when the AIS app starts regardless if the ADS-B app is run first.

Anyways I would say that the AIS app should be configuring the receiver modulation type to NFM instead of allowing the setting to be carried over from the last app, or depending on the modulation setting to be loaded from the rx_ais.ini settings file.

yellowz1234, 2023-07-03:
Thank you @Brumi-2021 and @NotherNgineer for looking into this issue.
After pressing the DFU button twice and looking at the receiver settings, I can confirm the behavior that @NotherNgineer described: the frequency is set correctly, but the modulation type for AIS is set to 3 after running ADS-B instead of 1. This causes the problem.
The workaround described by @NotherNgineer to enable save and load app settings after boot seems to work, but it would be better for AIS to always use NFM instead of carrying over the setting from previous app. Thanks!

Brumi-2021, 2023-07-03:
Hi @yellowz1234 , yes , as @NotherNgineer found , that is the reason of that problem.

Now we are discussing through discord with @kallanreed  how to implement a permanent solution.  

Soon we will fix , thanks

NotherNgineer, 2023-07-03:
Looks like the problem is not the modulation per se, but the tuning frequency offset which is different when modulation is set to "Spec" mode.  PR #1235 should resolve the issue.

NotherNgineer, 2023-07-05:
Actually the fix was in PR #1236 and I withdrew PR #1235.  Please confirm that the nightly build n_230705 resolves the issue.

I'm closing the issue for now believing it to be resolved, but please re-open if the problem still exists.

closed 2023-07-05 by NotherNgineer

yellowz1234, 2023-07-06:
Tested with the nightly build n_230706 and can confirm that the issue has been resolved.
Thanks!