portapack-mayhem/mayhem-firmware #1238: Splash screen switcher

sizzlik, 2023-07-05:
This idea was mentioned on discord before and maybe can be realized someday when someone finds time. Not important but "nice to have" feature:
Some sort of app that lets you view and switch splash screens. Most features are already in the code, view an image, open/move/copy files. So you can put all your favourite splash screens in a /splash folder, view them with the image viewer and on a button press the selected image gets renamed to splash.bmp and copied into root, overwriting the current one. Right now i got 4 splashes in root named splash1, splash2...and rename them with the file manager when i get bored of the old one. Having that a bit automated would be a nice QoL addition

eried, 2023-07-05:
It is the perfect idea for implementing as first PR 😁 hint hint, just saying

NotherNgineer, 2023-07-05:
Regarding "image viewer", I find it interesting that File Manager doesn't display a BMP file when it's clicked on (as it does with the PNG screenshot files).  I bet it would be easy to add support to display a BMP file when clicked upon, given that there's already support for displaying the BMP as a splash screen.  Maybe someone will implement that.

Also there's probably room for a few more action icons such as a "Set As Splash Screen" button in File Manager when viewing the file list, if we could figure out how to represent that action in a 16x16 monochrome pixel button (maybe a tiny arrow pointing to the screen of a tiny PortaPack?; these are the kind of tough decisions we're up against :-) ).

NotherNgineer, 2023-07-05:
@sizzlik Please give the attached test version of firmware a try.  Use File Manager to navigate to the directory where you have Splash screen images and click on each of the BMP files to view them.  If you press the "right" button when viewing an image, it will be set as your Splash screen (replacing any /splash.bmp file that you might have already had).

[portapack-h1_h2-mayhem.bin.zip](https://github.com/eried/portapack-mayhem/files/11960979/portapack-h1_h2-mayhem.bin.zip)

sizzlik, 2023-07-05:
> @sizzlik Please give the attached test version of firmware a try. Use File Manager to navigate to the directory where you have Splash screen images and click on each of the BMP files to view them. If you press the "right" button when viewing an image, it will be set as your Splash screen (it will be copied over any /splash.bmp file that you might have already had).
> 
> [portapack-h1_h2-mayhem.bin.zip](https://github.com/eried/portapack-mayhem/files/11960979/portapack-h1_h2-mayhem.bin.zip)

89e8956d tested and works like a charm. Went through 5 bitmaps and back with no hiccups. Tested with a not splash sized bmp and it gives the right info "Not a valid splash image". Thumbs up from me a PR candidate. Thanks for your work.

NotherNgineer, 2023-07-05:
The changes in this test-code will be included in nightly build n_230706, so I'm going ahead and closing this issue.  Feel free to re-open if you find any issues.  Thanks.

closed 2023-07-05 by NotherNgineer