portapack-mayhem/mayhem-firmware #1248: Worldmap seperate download

loewal, 2023-07-07:
The sd-card files contain the worldmap, a quite large file.
This has to be downloaded only once...at leat I think this will not change very often.
Is it possible to make that a seperate download and describe that it has to be copied into the adsb-folder if not already present?
In this way it is not necessary to download a 450 MB every time one wants to get a fresh copy of the sd-card contents.

jLynx, 2023-07-09:
This is what we previously had to do and had lots of users complaining that it was missing which is why we sent a lot of time trying to get it Incorporated. So won't be removing it sorry

jLynx, 2023-07-09:
However, we could however have a 2nd download that doesn't include it. So mayhem_v1.7.3_COPY_TO_SDCARD-NO-WORLD-MAP.zip

jLynx, 2023-07-10:
This will be available in the next nightly and stable releases from now on

closed 2023-07-10 by jLynx