portapack-mayhem/mayhem-firmware #1252: Triggering recording in “Capture”

trezorian, 2023-07-08:
In the "Capture" function, it will be useful to be able to trigger recording after reaching a set level (something like Squelch or VOX) after the signal disappears, waiting a few seconds and saving the file or stopping recording until the next exceedance of the set level.

This will allow you to record rare and short signals (e.g. remote controls, weather stations or sensors with a radio interface) without waiting for them.

Regards, Krzysiek

gullradriel, 2023-07-08:
While this can't be done in capture yet, you can do this in Recon.

gullradriel, 2023-07-08:
It's working like you said, trigger by a squelch level, record, stop after X milliseconds of inactivity

trezorian, 2023-07-08:
You're right, that's what I meant.
Thank you.

closed 2023-07-08 by trezorian