portapack-mayhem/mayhem-firmware #1254: Make error

Naster17, 2023-07-08:
[100%] Built target application.elf
[100%] Generating application.bin
[100%] Built target application
[100%] Generating portapack-h1_h2-mayhem.bin
Traceback (most recent call last):
  File "/home/nik/portapack-mayhem/firmware/tools/make_spi_image.py", line 78, in <module>
    raise RuntimeError('SPI flash image size of %d exceeds device size of %d bytes' % (len(spi_image), spi_size))
RuntimeError: SPI flash image size of 1059460 exceeds device size of 1048576 bytes
make[2]: *** [firmware/CMakeFiles/firmware.dir/build.make:75: firmware/portapack-h1_h2-mayhem.bin] Error 1
make[1]: *** [CMakeFiles/Makefile2:670: firmware/CMakeFiles/firmware.dir/all] Error 2
make: *** [Makefile:101: all] Error 2

zxkmm, 2023-07-09:
Lol but isn't the error msg already said, "SPI flash image size of 1059460 exceeds device size of 1048576 bytes"? So what's the question?

gullradriel, 2023-07-09:
You're not using the right compiler.
Without hacks the classic build will only work fine on referenced compiler version.
Hence the size is too big.
Follow the arm debian instructions build in the wiki, adapt them to your needs.

closed 2023-07-14 by Naster17