portapack-mayhem/mayhem-firmware #1262: Problema Ricezione solo radio locali

kewinxx, 2023-07-11:
Saluti  a tutti ho acquistato questo HackRF

https://it.aliexpress.com/item/4000247041639.html?spm=a2g0o.order_list.order_list_main.35.21ef3696TMxrHm&gatewayAdapt=glo2ita

aggiornamento fatto  Mayhem firmware v1.7.3

ma non riceve  vhf- am ecc eecc  riceve solo radio locali provato con sdr sharp
ce qualche impostazione da fare ^????

Brumi-2021, 2023-07-11:
Hi @kewinxx , try to play with the game parameters .
I think sdrsharp initially has a Vert low gain and the receiver looks like damaged , 
LNA more than 30,
IF VGA 20 to 30 
, AMD AMP 1 for weak signal . But check no RF saturation . 

[image]

kewinxx, 2023-07-12:
ok grazie gentilissimo  ma il problema lo fa anche senza sdr sharp

jLynx, 2023-08-07:
Please keep the issues in English so we can all understand and help

closed 2023-08-07 by jLynx