portapack-mayhem/mayhem-firmware #1273: Tuning knob just toggles frequency up and down - 2 different units - see below

chris451gh, 2023-07-14:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly version here: https://github.com/eried/portapack-mayhem/releases)
*****Note I have tried several releases of the hackrf mayhem firmwares the knob behaves the same. Sometimes to tune I can do it by pressing sideways on the knob in a northwest direction. I have read about other users who have this problem. It may relate to the brand of rotary switch or it may be common to all units but I only have 2 to look at and both of them were from different sources and they behave the same. There are videos on youtube showing how it works....

----
Problem: tuning knob doesnt work properly. It toggles between two frequencies.
The frequency is hard to move up and down, sometimes dialing fast works.
Easier to type in the frequency than to use the dial. 
The bug: Hardware design is to pull down the switch lines using a pull up resistor.
Switch one and switch two are out of phase. The pulses do not overlap. So when 
the switch closes, the signal line is pulled down, the logic can't tell if its an up or down, 
interprets the non overlapping pulses as up and then down up and then down.
Possible fix: invert the logic in software so there is overlap. Then the direction of rotation can be found by simple logic.
The 'phase' can not be interpreted unless the pulses overlap, but with pull down logic
there is no overlap. If the hardware was pull up logic then the pulses would overlap at some points. Rather than connecting 3.3 to the common and putting a low value pulldown resistor, which would be a hardware mod, maybe inverting the signal logic in software would work. Not sure if this is interrupt driven or bit banging to read the two lines from the rotary switch or if there could be a different logic used to solve this problem. A different type of switch may have overlap like this

A software fix to a hardware design problem.
Note: after the rotary switch is used for awhile it wears slightly and the pulses in pull down logic do not overlap. Then the behavior is when the knob is rotated continuously the frequency just toggles up and down by one step whatever the tuning step wsa set to.
If the logic was inverted then there is overlap and it may work but be opposite direction,
but then the expected behavior is turning in one directon continuously will add steps in one direction and subtract steps on the other direction of rotation.
Scope image attached.
**Describe the bug**
A clear and concise description of what the bug is. 
Tuning with knob goes up and down instead of continous up when knob is turned clockwise
Tuning with knob goes up and down instead of continuous down when knob is turned counterclockwise.
Tuning decrements when the knob is turned clockwise very fast, then it is hard to stop at the target frequency. 
**To Reproduce**
Steps to reproduce the behavior:
1. Go to '.audio receive and select frequency with buttons, rotate tuning knob..'
2. Tap on '....'
3. Tune very fast, freq will tune opposite the expected direction. Scope shows more of a shark fin waveform, and there is some overlap.

**Expected behavior**
A clear and concise description of what you expected to happen.
Freq tunes up in clockwise rotation
Freq tunes down in counterclockwise rotation
**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release: 
* Latest Nightly release:n230702
* Previous working release: None.

**Additional**
line 1  01100110011001100
line 2  00110011001100110
What you have
line 1  0001110001110001110
line 2  0110000110000110000
Now invert what you have and it goes to overlap.
Possible problem: A new rotary knob switch may have overlap until its used for a while in the reverse logic. 

Other notes:
The rotary knob needs a spacer placed at the top of the hole in the rotary knob so
that pressing in on it can be used to turn the unit on/off. No documentation for this mechanical problem, if a hackrf one is retrofitted with a portpack H2 board, then a wad of paper in the hole works to space it up from the cover enogh that press to turn on, works.
Also so that press twice to turn off works.

If the bug is difficult to explain, additionally to the text please include images and videos.
[image]

gullradriel, 2023-07-14:
Have you tried to change the encoder mode in 'Settings' ?

chris451gh, 2023-07-14:
In that setting you have sensitivity low normal high, those change the rates accordingly but the freq toggling behavior remains.
I turn it very fast in the opposite of the desired increment and this makes the wave form start to merge together and it will go in a continuos direction.  If I turn slow then it toggles. Its because the square wave rounds off in the leading edge to the point where it overlaps the pulse on the other channel.
A new switch with no wear on it seems to work correctly, so I used that unit quite a bit and before long the toggling behavior began on that unit. Should I try the hardware fix I think will work, invert the signals and try it? -Chris

NotherNgineer, 2023-07-15:
Thank you for reporting this issue.

To answer one of your questions above, firmware polls the encoder's Phase A and B bits (not interrupt driven).

The two encoder phase bits are visible in Debug->ButtonsTest (turn the dial very slowly to see them).  To confirm that software is seeing the same as your scope, it may help if you could you provide a short video showing how the phase bits change as you turn the dial very slowly in each direction.

Firmware is indeed expecting the encoder bits to transition as you've indicated under "Additional" above, i.e. 00 -> 01 -> 11 -> 10 -> 00, or in the reverse order if turned in the opposite direction.  With your encoders, it seems that it is skipping over the 11 "overlapped" state and thus it's more like this:  00 -> 01 -> 10 -> 00.  If you invert both signals, firmware would see 11 -> 10 -> 01 -> 11, but now it's missing the 00 state.  Since current firmware does not understand either of those transition sequences, I don't see any need to make the experimental hardware changes.

In the next day or two, I will attach a test version of firmware that accepts this bit transition sequence.  I have assigned this issue to myself.

NotherNgineer, 2023-07-15:
A test version of firmware is attached.  You may need to adjust the encoder dial sensitivity setting under settings.  Please let me know if it resolves the issue.

[portapack-h1_h2-mayhem.bin.zip](https://github.com/eried/portapack-mayhem/files/12062412/portapack-h1_h2-mayhem.bin.zip)

chris451gh, 2023-07-15:
I will try it also will upload some videos and pics.

chris451gh, 2023-07-16:
Thanks very much for the test build. The result was still about the same, rotating the knob would increment up then down on each detent. I am convinced the problem on my boards is the rotary encoders. Connecting a scope to the signal pins they display a 180 degree phase offset, not a 90 degree phase offset. CW VS CCW look like they would read the same.  It is a fluke that they work at all. These parts came installed on the portapack board, but I have not identified the manufacturer.  The spec for the PEC part I found on Mouser shows a waveform in which the phase offset is 90 degrees. The wavforms get close enough to work sometimes with side pressure on the shaft of these encoders. If you have a portapack H2 board look at the markings on the rotary encoder and report back - what is the part id if  shown any letters on it and how does it work? The top of the encoder which is causing problems is "FRE"
 I am looking into replacing the encoder on the portapack board with this device:
PEC11S-9215K-S0015, I checked the dimensions vs what is on the board. This part will fit into place. The pin connections are the same.  It has the press in switch and this version has the knurled knob. The part on the board is labeled PRE and this replacement is Bournes, you can  see it on Mouser.com or search for it.  
I will order this part and install it, and test the portapack tuning feature for correct operation.
I will report back if this fixes the issue. Expected by 7/20/2023.

NotherNgineer, 2023-07-16:
Yes it will be difficult for firmware to determine the direction if the encoder outputs just alternate between 01 and 10. 
My encoder says "PRE" on the top, but I can't tell the part number.  I look forward to hearing the results of your testing with the replacement encoder.

chris451gh, 2023-07-20:
I changed the encoder to the bournes PEC11S-9215K-S0015. Removing the old encoder required great care so as not to pull up a pad, use solder wick on the terminals to remove as much solder as possible, then use tweezers and an iron to lift each lead, repeat for 5 angled leads. Then heat one mount tab and lift, remove solder as needed.
Prepare the pads by removing any solder.
Place the part in place, remove indexing posts(Plastic from the back) if it does not go in the alignment holes on the board. Solder 7 places.
Tested fine:
Each detent increments or decrements the parameter selected precisely, so if you are having problems with the rotary knob replace the encoder with this. Note: remove any stuffing paper that may be in the knob center hole. Height is above nominal so if a flush knob is desired you have to remove approx .050" from the top of the encoder shaft, for this part number.  Old part: 180 degrees out of phase(why) New part: 90 degrees out of phase for each switching pulse. Detents correspond with one increment or decrement. There is no sign of the ambiguous behavior 'back and forth'.

chris451gh, 2023-07-20:
Change your part if labelled "PRE" and your tuning knob will work correctly.

closed 2023-07-20 by chris451gh