portapack-mayhem/mayhem-firmware #1274: Virus Alert (hackrf_update.exe)

FTHKDG, 2023-07-15:
File : hackrf_update.exe
Antivirus : MaxSecure
Result : Trojan.Malware.300983.susgen
Virus Total Link :  https://www.virustotal.com/gui/file/6d73d84840c7804d25d741c294b33ecc00552d83435a4335e6ce1b8ff6af61a1/detection

Why am I getting this warning, is it reliable?

jLynx, 2023-07-15:
No idea why that is showing up, but you are able to just ignore it. This file has never changed. Assume you got it from this repo, then no need to worry

sizzlik, 2023-07-15:
if you got your portapack as preinstalled/preassembled kit, you dont even need the exe i think. Just pop the firmware in the folder and use the inbuild firmware flasher.
The AV Alert is a false positive. See the name "susgen". Thats a tool used for heuristic analysis. Means it looks into the giblets of an element and tries to categorize its behaviour based on its functions/api calls/interupts/etc ..before any code is executed. (Usually AV's only look for the V-Signature of already known threats.) And guestimating the intent of a program, by judging what tools it got on its belt, is prone to false positives. The functions used in the exe to flash the chip are pretty much the same tools a trojan would/could use to gain hardware persistence, eff up your Bios and all the funsies. If i got a hammer on my belt, am i a burglar or a carpenter? AV is better save then sorry and locks up.
If you click details on virustotal you can see the indicators picked up..its a PE-exe (Portable Executeable), those are always sus to AV's, plus importing kernel32 and SETUPAPI..that turns the little susgen into an TSA Agent. Basically it sees a standalone, unsigned executeable with the capabilities of probing your system and reprogramm any hardware with whatever code the author wants. And thats exactly what it does..probe the hRF for its HW, run a few checks and then write code into hardware (or not)..you would not want such programs on your system if the program claims to be just a screensaver (.scr files are secretly PE-exe but who gives..) hackRF_updater.exe ist just a tiny harmless tool, misunderstood and scared it resides next to always renewd code and waits for its time to rise and shine and again make someones day, like it did mine back then. Thank you, hackrf_update.exe.

(Guess i got a little carried away there..)

jLynx, 2023-07-15:
Either way, you summarised it well, and yes you can put the bin file into the firmware folder and flash it from there

closed 2023-07-15 by jLynx

FTHKDG, 2023-07-15:
> Either way, you summarised it well, and yes you can put the bin file into the firmware folder and flash it from there

thank you for the detailed answer.