portapack-mayhem/mayhem-firmware #1277: Capture mode does not work properly for lower bandwidths and uses 16-bit IQ files unnecessarily

voxo22, 2023-07-16:
Hello,
I've tried all the recording modes for Capture mode, but only 250k and 500k are actually usable. For the modes that use smaller bandwidths, there is some kind of weird frequency shift, the source cannot be found where it is actually transmitting. The smaller the bandwidth is set, the lower the source appears to be in frequency. Recording in <250kHz modes is unusable, replay just doesn't work.

Mayhem Capture uses 16-bit IQ recordings, while hackrf hardware only supports 8-bit IQ samples (as most of SDRs do, check with hackrf_transfer). So this unnecessarily doubles the required bitrate, card space and slows down SD card recording that could be sufficient for >1M bandwidth.

Can you please check it?

NotherNgineer, 2023-07-18:
I don't know the history behind this, but I concur with your findings that upper bytes in all of the C16 files are unused (bits 8-15 all match the sign bit 7).

It will not be easy to support 8-bit IQ internally, since all code I see is expecting 16-bit IQ (likely all the way back to the CPLD, even though the HackRF is 8-bit, as you mentioned).

One thing I can do is implement a simple conversion function to optionally read & write capture files as 8-bit IQ for possibly increased capture bandwidth (SD card write latency has indeed been an issue).  I'll have some test code tomorrow to experiment with that.

voxo22, 2023-07-18:
Great! While external hackrf utilities support mainly 8-bit files, so the records from the card could be easily parsed and used there and vice versa without any conversion. Especially if they could be in 8-bit signed format.

Brumi-2021, 2023-07-18:
Hi , current Capture is using a fixed decimation of 8x and that has many benefits (and M4 resources), in terms of using less ideal filters, avoiding DC spike with off-center DC capture , (allowing to use more simple and fast filters , with wider  stop transition band ).
And in fact , I already discussed it long time ago with original guru designer Sharebrained,and he clearly told me that thanks to that , although original ADC sampling is 8 bits , that full decimation and full digital process ,is actually  expanding theoretically the captured digital dynamic marging around 2 bits , so in fact we got equivalent recording to around 10 bits . (So it is true that we are not using the full .C16 , but 10 is better than just 8) .

Anyway , you are pointing that lower bandwidth capture is not working well . (I do not know , I just tried deeply 100k , 250k and 500k and to me they seems to work well .) How did you confirm that is not working correctly ? Can you clarify more how did you test it ? Which recorded signals ?. Did you recorded some specific NBFM ? 

 (Note aside : Anyway in order to capture in future correctly 1MHz and 2Mhz , there are pending some future TODO ideas, with  other  strategy to avoid current M4 oversaturation and dropped samples, reducing decimation or recording directly in .C8)  , but I do not suggest to do it in current below bw captures <=500 Khz )

gullradriel, 2023-07-18:
Had no problems capturing and replaying using latest nightly.
@voxo22 what version are you using ? There were some with modulation troubles at some points.
Is your TXCO correctly set ?

voxo22, 2023-07-18:
ok, below are some snapshots. Left - mayhem v1.5.3, right - mayhem 1.7.3. I have tried to capture keyfob for simple control of windows blinds (314.375 MHz) on v1.7.3, I was unable to see the signal in capture window with correct freq while using <25k BW. It is shifted mostly downscale, e.g. with 8k5 BW I can see the signal in frequency about 200 kHz lower compared to real (433.92 MHz can be found somewhere around 433.705 MHz). On both version there is different speed of waterfalls e.g. between 8k5, 16k (normal) and 11k (fast), is this normal?

While using any mode <250 kHz I was unable to replay the signal, windows blinds does not "understand" the transmitted code, does not matter if signal is recorded weak, strong or normal (blinds do not use rolling code). Opposite of that, 250k, 500k and 600k worked ok, 1M is somehow "chopped", but this is known.

[image]
[image]
[image]
[image]
[image]
[image]

gullradriel, 2023-07-18:
Are you using the same antenna ?
What version of hackrf do you have ?
Are you CPLD up to date ?
Do you have a coin battery installed, is it working ?
Are you using App Settings options ?
Have you tried switching antennas ?

Brumi-2021, 2023-07-18:
Can you show the same tuned commercial FM broadcasting in capture 500 Khz , in both fw versions at the same time ? 

And also later from the Audio Rx WFM , do the same ,

thanks

Brumi-2021, 2023-07-18:
Hello @voxo22 , ok ,now  I better understand you.
Your point is that we are not sure if 8k, 11k5 , 16k capture bandwidth are working well . 

You could be right , I never checked in detail  nor used those bandwidth , but I had confidence about 100k, 250k and 500k .  We need to check it . 

anyway I understand that the bandwidth of your  tuned AM signal has higher bandwidth than the DSB 9k filter , not ? (Maybe 10 to 12 Khz total Bandwith , then we should be able to record using  16k capture (if no bug)

voxo22, 2023-07-18:
Hello @Brumi-2021 , yes exactly. When I tested 25k and 50k BW, I may have used smaller BW, that is normally required for correct decoding. I understand this. But I do not understand that weird center freq shifting. I assume, that narrowing capture mode BW should "zoom" the waterfall picture and I should see wider "red" zone - active signal or carrier. It should never shift center freq to any side, if it is properly tuned.
It does not really matter, which BW to choose, if we want to record any AM signal, because narrowing BW will only narrow audio spectrum. So we should be able to select even 8k5 BW.

Brumi-2021, 2023-07-18:
I just added a new comment to the previous post.
I think you are right , there is something strange in 16k, 11k, 8k5 capture bandwith .

NotherNgineer, 2023-07-19:
Regarding code I've written to support for reading & writing files in .C8 format, I'm running into what looks like a cache coherency issue when the ARM CPU is accessing the buffer memory, so I'm sorting that out...

NotherNgineer, 2023-07-20:
Attached is a test version of firmware that supports Capture & Playback in C8 format.  Conversion is being done at the point where data is being read or written from the file.  It doesn't look like the sample rate can be increased much beyond where it is now, but probably it could be optimized.  The least significant byte is being stripped off when converting C16 to C8, so a strong signal is recommended.

[portapack-h1_h2-mayhem.bin.zip](https://github.com/eried/portapack-mayhem/files/12113610/portapack-h1_h2-mayhem.bin.zip)

(Note that this doesn't do anything to help with the other issue with smaller BW settings.)

chris451gh, 2023-07-23:
I have been analyzing the capture playback modes. I reproduce the settings such as sample rate with SDRAngel, convert the c16 files to .sdriq with iqcv.py, and play them with SDRAngel and they look OK to me, other than the frequency shift that occurs on recording at the low bandwidths and the same shift on replayed files they work for me. The 'freq error' is an artifact of decimation and integration factors.
These are visible and can be changed in SDRAngel. 
 I came to some conclusions.
The capture mode could be emulated with a hackrf controlled by SDRAngel.. This local osc is mixed and sampled at the 'main sample rate' in this case I found 3200000 to be a good setting to simulate the mayhem firmware. From that sample rate there is the first decimation factor, 4, then after the dacs another decimation factor of 8 takes 800khz down to 100khz in the fpga 'hardware'. This IQ data is lowpassed and stored in the file.  It works about the same when the 1st decimation is 2 and the 2nd decimation is 16. 

SDRAngel has transmit ability with the converter block setting to integrate the converter block, (4), integrate the IQ block (8), and it tells the offset depending on the integration number.
int   frq offset
8      -350000
4       -300000
2       -200000
1       -0 hz.
In the mayhem firmware at the low sample rates it was a little different.

The hackRF mayhem firmware does not tell us the 1st converter decimation/integraion factor.
It does not tell us what the IQ block decimation/integration factor is(power of 2)
There is a gain adjustment on SDRAngel. It goes from -10 to +50 db (multiply complex number by the scalar value of this db).
There is a peak and average metric showing when there is clipping (you clip you get noise)
You clip infrequently you get pops of noise, I watched the spectrum and cranked this around, very useful. Would be helpful to adjust gain of playback. It would be useful to have calibration controls to apply a fixed dc offset to the DACS in transmit, to minimize carrier and maximize signal that is being replayed. There is not one in SDRAngel either, but a dc offset could be added to an IQ data file to find the calibration factors needed to null out the mixer to minimize carrier leakage.
Capture mode sample rate 25k still centered up?direct sample? , 16k its offset by -62khz(decimate 2), 11k sample is offset by -200khz, 8k5 offset by -250k,  
Would be helpful to see what the first and second decimation/integration factors are, 
to know what the mix offset is.
Offseting the recorded carrier to be in one sideband seems to be a must.
The carrier, mixing, and intermod (too few bits?) seem to trash a playback signal thats centered up. IF its a strong signal captured on one side then played back on one side its good, but the higher the sample rate the better because it reduces carrier to signal ratio. I don't know if thats a mixer rolling off near 0 hz or capacitive coupling in ADC DAC but its the same problem in Hackrf firmware or SDRAngel settings. If the ADC or DAC are working close to the 'dc point' there is a rolloff of the signal so that the carrier leak is much stronger than the signal, and its jamming a centered signal.
However an FM broadcast captured at 500k, offset to the right side or even centered, seems to work without a problem and sounds great on RF capture playback.

Suggestions:
display the decimations in the ADC and the DSP block, and calculate the correct tuning frequency as it walks downward as the decimation is increasted and the bandwidth is decreased. 
What is the factor, for each power of 2 decimation it walks down by twice the sample rate?
Use the SSB blocks to capture a wider bandwidth, at least 24 khz upper sideband would  allow a channel to be captured and played back with the benefit of adjustable filtering gain control and IQ fidelity..
Currently the hackRF record plaback of ssb is limited at 3khz. Is it possible to increase that to 24khz? Save as wave file at 48khz? Stereo for I&Q similar to the SDRSharp wave files?
If that w
[image]
[image]
ere possible a digital signal of less than that bandwidth could be recorded and played back in the audio realm into an ssb modulator and should work great.

Brumi-2021, 2023-07-28:
Hello @voxo22 and @chris451gh  thanks for your investigation and explanations.
All of us agree that current Capture App , only is usefull from BW : 25k , 50K , 100K , 250K , 500K  (to record correctly) .

So I have checked and currently  8K5 / 11K / 16K  are just not usefull and not correct.  (shift center , and strange signal ) .
I think  total system did not work well  with ADC sampling below 200khz .  (we need to increase it in those lower BW options ) . 

I have been also  investigating it , and so far , my current status , 
1-) I got a nice  new proposal settings for the lower BW : 8k5 / 11k / 16k and they are working quite well in the screen.
2-) But I still need to solve the decimation in the Write SD card part . 

So the investigation is not finished , but just to show you the current status.

FYI : 
Some example pictures of Capture App screen , 
using the  same Tx signal test : NBFM of our signal Generator , with deviation 3Khz   and modulation :  1khz tone . 
[image]

  
Pending ,  to solve the write decimation part , 
and then, code it ,  to make those settings dynamic .
DRAFT ,   TO BE CONTINUED.

Brumi-2021, 2023-07-29:
Hello again @voxo22 and @chris451gh  ,  (and all) 

Here you can have an advanced test bin ,  of fixed settings to all BW options ,  to allow good lower files captures , from 12k5, 16k , 20k onwards.   

The anti-aliasing filters are still not optimised,  but at the moment , I think that 
        the recording capture screen and axis x freq. scale is correct.
        now there is not freq. center shift. (it is correct) 
        the recorded file size and playing secs  is matching with the selected bitrate, and its metadata.

But I was getting crazy, when recording different BW , and always in the replay we got the same x axis freq. scale. 
I think  the Replay App we have another small bug, (always the display axis freq. is the same , not  adaptive to the different sample rates of the metadata) and that was confusing me. ) but this is different discussion . 

That binary is  not finalized,  it is just optimizing the settings for the lower bit rates,  12k5 ,  16k , 20k .  (but just to test,  now it is applying those settings to all BW options . But the idea would be dynamic combining previous settings + those ones   ) 

To be continued, 

[portapack-h1_h2-mayhem_test_Low_CAPTURE.zip](https://github.com/eried/portapack-mayhem/files/12207595/portapack-h1_h2-mayhem_test_Low_CAPTURE.zip)

closed 2023-07-31 by Brumi-2021

chris451gh, 2023-07-31:
I will give the build a try and report back to you. Thanks!

Brumi-2021, 2023-07-31:
Hello @chris451gh , thanks 
Better pick up tomorrow’s nightly fw , it will have integrated that PR #1332or pick up it from test drive discord .