portapack-mayhem/mayhem-firmware #1283: Capture App hangs if it's the first app executed after power-up

NotherNgineer, 2023-07-18:
I've noticed in the latest nightly builds that the Capture app will hang if it's the first app executed after power-up.  If another receive app such as Audio is executed first, then the Capture app will work fine.

I'll be looking into this, but wanted to get a bug report started for tracking purposes.  There may be a dependency on App Settings in some firmware versions, and perhaps also a dependency on the power-up state of some uninitialized RAM, TBD.

Affected nightly firmware versions include n_230712 through n_230718, and probably older, TBD.

gullradriel, 2023-07-18:
Capture starts fine just after a boot on my H2 no coin batt pmem from/to sdcard and H1 with working coin batt, no pmem to sd, no app settings load/save used on both devices.
I'll try on my H2 without restoring pmem (so it'll act as a fresh/reset install) and keep you informed.

Brumi-2021, 2023-07-18:
Hi @NotherNgineer I also could it in some  local compile binary , but not recently . And i agree with you , that in that case,  if previously you open looking glass or audio App or any receiver it works well.  
But let me check today's nightly.

Brumi-2021, 2023-07-18:
Hi again @NotherNgineer ,  I found one way to reproduce it. 
if I pre-flash an old version first , example 1.43  and later go to our recent yesterday binary  ,in next boot,  it happens. 

But if you boot , go prior to Search App open it and exit , and then comeback to Capture once,  it is fixed. 
  (I do not know why , but  only doing once , it  is fixed  for any new boot) .

NotherNgineer, 2023-07-18:
I can reproduce it with every version I've tried so far.  Maybe if it's uninitialized memory it shows up better on mine for some reason.  My simple test is to flash using the Flash App, reset PMem to default, power cycle, run Capture app.  I'll track it down this afternoon.

NotherNgineer, 2023-07-19:
Testing different firmware versions, the first version I see with this issue is n_230705.

I subsequently found that if I put the line "receiver_model.set_modulation(ReceiverModel::Mode::Capture);" back into its original location in capture_app.cpp, instead of relying on RadioState to do it, that somehow it seems to resolve the issue.

We still need to understand why this helps (could be timing), but this is a clue!

NotherNgineer, 2023-07-19:
It does seem to be a timing condition IMO.  The issue goes away even if I put some other line in there in place of setting the modulation, such as DEBUG_LOG().

NotherNgineer, 2023-07-20:
Interestingly, I can make this issue worse if I run Replay and play back any Capture that was recorded at a higher bandwidth, e.g. 2MB/s.  Once I do that, I cannot run Capture again, even if I power cycle and despite having a delay after starting baseband, unless I run a receive app (such as Audio) first.

NotherNgineer, 2023-07-22:
PR #1285 and PR #1287 are two different workarounds for this issue.  Closing.

closed 2023-07-22 by NotherNgineer