portapack-mayhem/mayhem-firmware #1284: Tuning down below 800kHz

voxo22, 2023-07-18:
Hi, while using Audio feature, is there some reason why I cannot tune below 0.8 MHz? HackRF can be theoretically tuned down to 0 kHz, but with Mayhem/Portapack cannot. Is it locked in software, or is it just Portapack hw limitation?

gullradriel, 2023-07-18:
You 'can' tune to 0.8 but do not expect good results. Maybe you can not go down because of the step you use. You can just click on the frequency and try to enter it manually. 
That said, HackRF is sold for 1 MHz to 6 GHz operating frequency (see GSG page [here](https://greatscottgadgets.com/hackrf/one/))
To use low frequencies like this you have to use something like an up converter (a hamitup or equal).
For the WFM decoding, I don't know for the jack. Not sure we even have enough CPU power to process the stereo.

gullradriel, 2023-07-18:
You may ask the remaining question in a dedicated issue.
It's always better to make one question by issues, except if they are really related.

closed 2023-07-18 by gullradriel