portapack-mayhem/mayhem-firmware #1297: Backup battery dies in a month or two high current mode found.

chris451gh, 2023-07-23:
----
The portapack clock and current state seem to be lost upon power glitch, power off, etc.
Bugging the battery it was at 200mV.
I connected a dc meter to the battery socket and connected a 3v supply, to measure the current.
The CR1220 has a capacity of 40mah.
When the hackrf is powered on the draw was 0.8uA.
When the power was shut off with two clicks of the button, the current draw was 68.7uA. If I toggle it slow or randomly after a few tries oit goes into a mode where it draws 2.7uA.
In the first case thats only 582 hours of standby.
In the second case thats 1.7 years of standby. 
Something on the board has two modes, one is higher current and one is lower. Looking for the schematic of the RTC circuit.
I saw a portapack h1 schematic but someone said the h2 was a modified verson that the documentation was never uploaded for(china versions?)
Looking for the Kicad .sch of it, I had a schematic/board layout but lost it to a corrupt hard drive. Need to find it again.

(Please try the latest nightly release before submitting this. You can find the latest nightly version here: https://github.com/eried/portapack-mayhem/releases)

----

**Describe the bug**
A clear and concise description of what the bug is. 

**To Reproduce**
Steps to reproduce the behavior:
Take apart portapack H2 to the point where any external chassis is removed.
Pop out the coin battery from the side, by pressing the gold contact away from battery, it pops out.
That is the positive, connect a 3v supply in series with a current meter set to milliamp range. 
Press the on button, get low current 0.8mA.
Press the on button twice slowly to the point where it just shuts off, longer pause. Standbuy current is high, 0.068mA. or low, 0.0027mA, somewhat randomly but less onften high if the button push twice is slow 0.5 seconds or more. 
Need to look at the rtc chip datasheet, and the schematic as to what is connected to the backup battery. Could be a mode the RTC chip goes into, and maybe has registers that are set wrong. 

1. Go to '...'
2. Tap on '....'
3. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release:
* Latest Nightly release:
* Previous working release:

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.

NotherNgineer, 2023-07-24:
I'd love to see an H2 schematic as well, but all I've seen are H1.

As far as I know (from the older PortaPack H1 schematic and old HackRF One schematic), VBAT goes from coin cell on PortaPack through a header connector to the HackRF One board (pin 1 on header P20), where it connects to pin 127 on the HackRF's NXP LPC4320FBD144 processor, which has an internal RTC and some persistent memory.  There may be some tiny capacitors in parallel with the coin cell on the PortaPack (C41 on H1 boards) and on the HackRF (C136).

Is there any current drain when the HackRF board is detached from the PortaPack board?  This should reveal if the PortaPack is drawing any power from the battery itself.

Maybe there's some latched up gates in the NXP CPU or something if the main battery power is not shut off cleanly or something (just guessing here).

NotherNgineer, 2023-07-24:
I-BAT should be about 3 uA typical according to the NXP LPC4320 specs, so 2.7 uA sounds about right under normal conditions.  I doubt there's any software configuration setting we could change to reduce it.  It would be interesting to find out if VBAT is connected to anything else.

MattPayne76, 2023-07-24:
Hi,

Same problem for me.
Portapack H2 from OpenSDRLab.
Backup battery (CR1220) died, settings and date&time no more saved after powering off.
I have no clue about the reason.
Also the main battery is also emptying while device is off.
[image]
[image]
[image]

eried, 2023-07-24:
> When the power was shut off with two clicks of the button, the current draw was 68.7uA. If I toggle it slow or randomly after a few tries oit goes into a mode where it draws 2.7uA.

Nice research. There is a lot of custom implementations of the poweroff, we need to test this on more units. My first portapack powered off holding the middle knob down or center button. Latest I bought shuts down with double click only on the knob. 

I imagine 1.7 years (or at least >6 months) it is ok time. But I dont think this is related with the firmware in this stage

chris451gh, 2023-07-31:
Furthur research: Reference h1 schematic pdf.
There is a CPLD chip, into which the 'power button' goes...select button on the knob by pushing in. A line leading into that chip toggles power on and off. The low power state of that chip is listed as 27uA about what I get but sometimes I get just the RTC current. The RTC is in the CPU on the HackRFone. 
The CPLD is programmable via quartus.  The code for it is 
\portapack-mayhem-1.5.4\hardware\portapack_h1\cpld
The freeware quartus tool may be able to open that CPLD source code.
The original H1 board had a cr2032 battery 200mah, the H2 has the 1220 cell which is is only 40mah. There was a change from Hold long to shut off before, and tap twice to turn off, later version.
I have not tested the current draw from the battery on/off but next time I open it up I'll test that, one person says the battery runs down when its off. 
The battery pack chip would disconnect the battery in the event of low voltage, when I got mine it had a dead battery and the display does not work until its powered up with a button held down, rather confusing to start it up, but in an hour or two the battery is charged again.
Current workaround: Take out coin cell to store. Plug it in and turn it on (blank screen) Operational battery: The management chip will disconnect it when its run down. Or open unit up and disconnect plug.
Two factors, a smaller RTC battery was substituted, and cpld code was changed in the CPLD, from Tap/Longtap to one Tap/ two taps, for on/off.

NotherNgineer, 2023-09-29:
The coin cell battery in my H2R4 died too after about 5 months.  I'm considering whether I might be able to replace the CR1220 battery holder with a CR2032 holder (if I can fit it), or connect the RTC to the LiPo instead (of course I wouldn't want to drain the LiPo down to 0V either or that's a bigger issue).

gullradriel, 2023-10-11:
One of my units isn't having a coin battery and lives pretty well without it, just by using the PMem to sdcard option.
Only caveat: it can't keep a track of time.
Else if you exit the apps correctly with back button, settings should be saved and restored

NotherNgineer, 2023-10-24:
I was using my PortaPack with a dead coin cell for awhile (using the PMem to SD option) but I missed having the actual time.  I would never replace the CR1220 cell with another (1) because I don't want to stock an odd size coin battery, and (2) because it's just too dang small (per this issue).  Here's my workaround using the popular CR2032 and a battery holder from Amazon.  I can deal with having to replace the coin cell every year or two.

[image]
[image]