portapack-mayhem/mayhem-firmware #1305: Hello everybody! I am completely upset!

Sats301, 2023-07-25:
Hello everybody! I am completely upset! 

Completely same situation, but with one little difference...
1. All actions with FW/CPLD manually flashing - not working.
2. Can't check situation with 1.6.0 and lower FW (screen not working, buttons can't help).
3. I've bought 100 PCS of HackRF PortaPack H2+ for student education, and the part of education program - not available, because we cant discover lite spoofing...
4. .C8 folder generated by gps-sdr-sim pack.
5. SD cards no make sense.
6. But in HackRF with PC - is fine.

I am really not understand what I need to do.
Just look on photos...
[image]
Hacrf_info

[image]
Chipsets

[image]
GPS_Sim - completely bad!
[image]

HackRF mode - is ok
[image]
[image]

Friends could I ask you for a help?
Because this situation is a really nervous...

_Originally posted by @Sats301 in https://github.com/eried/portapack-mayhem/issues/1070#issuecomment-1648826316_

jLynx, 2023-07-25:
So you are saying you can't flash the latest firmware and are stuck on 1.6.0?

Sats301, 2023-07-25:
> So you are saying you can't flash the latest firmware and are stuck on 1.6.0?

No, I test with every latest FV from 1.7.0 (all nightly releases and stabile 1.7.1-1.7.3). 

I can’t test 1.6.0 and lower, because PortaPack  can’t activate screen :(

jLynx, 2023-07-25:
If it's working with the computer then it should still work with the PP. I'm not sure what the issue is here myself, hopefully someone else with gps experience could chime in.

I assume you have read the wiki? https://github.com/eried/portapack-mayhem/wiki/GPS-Sim

Sats301, 2023-07-25:
> If it's working with the computer then it should still work with the PP. I'm not sure what the issue is here myself, hopefully someone else with gps experience could chime in.
> 
> I assume you have read the wiki? https://github.com/eried/portapack-mayhem/wiki/GPS-Sim

Sure, I created really good dynamic paths and static points. On old portapacks everything is fine. But new batch… 100PCS it is really strange. 
from other side maybe I need adjust something, frequency maybe. 

If you want I can share I/Q (.bin) or .C8 with parameter .TXT folders

jLynx, 2023-07-25:
By any chance are your new HackRF's the R9 version?

NotherNgineer, 2023-07-25:
@Sats301  You said that GPS Sim is working on old PortaPacks.
Which model PortaPack did it work on, and does GPS Sim work with *any* version of firmware on the old PortPack, or did it stop working after a certain firmware version?

Sats301, 2023-07-25:
> I think I know the problem. A couple of questions first though:
> 
> 1. By any chance are your new HackRF's the R9 version?
> 2. Does the non working new ones work when done via the computer, or not at all?

1. I can’t check, HackRF_info shows (no name). 
can I see this on the board? 
2. From a PC - everything is fine.

jLynx, 2023-07-25:
> 1. I can’t check, HackRF_info shows (no name).
>    can I see this on the board?
> 2. From a PC - everything is fine.

Can you please upload a photo of the top of the HackRF (remove the PP)

Sats301, 2023-07-25:
> @Sats301 You said that GPS Sim is working on old PortaPacks.
> Which model PortaPack did it work on, and does GPS Sim work with _any_ version of firmware on the old PortPack, or did it stop working after a certain firmware version?

On “old” HackRF PortaPack H2+ with any FWversions everything is fine.
 I wait my working PortaPack from colleague for checking HackRF_info.

Sats301, 2023-07-25:
> Can you please upload a photo of the top of the HackRF (remove the PP)

Shure, please wait a 30-40 min

jLynx, 2023-07-25:
@Sats301 Ok, so if you try that portapack on a non R9 I think you will find it works perfectly. The issue is the TCXO design changed with the R9 and 99% if not all portapack have been built with their TCXO out of spec, which means it does not actually work and as far as I am aware, it's needed for gps spoofing. So sadly I don't think any of these will be able to work, not unless you get non R9's or unless you get new PP's with the correct TCXO.

Specifically the issue is the TCXO needs to output a peak-to-peak voltage of 3v3 but thy all seem to output around 1v or less. This means they are out of spec. There was a "bug" in the older HackRF versions (<R9) where it allows this 1v TCXO to work, but with the new design R9's this "bug" was fixed and now they no longer work.

HOWEVER, you said these r9's work on the computer? Can you 100% confirm that? Because I would not have expected that to work, and if they do then there is potentially hope. I'm just not sure how they are working with the current TCXO

Sats301, 2023-07-25:
> @Sats301 Ok, so if you try that portapack on a non R9 I think you will find it works perfectly. The issue is the TCXO design changed with the R9 and 99% if not all portapack have been built with their TCXO out of spec, which means it does not actually work and as far as I am aware. A working TCXO is needed for gps spoofing. So sadly I don't think any of these will be able to work, not unless you get non R9's or unless you get new PP's with the correct TCXO.
> 
> Specifically the issue is the TCXO needs to output a peak-to-peak voltage of 3v3 but thy all seem to output around 1v or less. This means they are out of spec. There was a "bug" in the older HackRF versions (<R9) where it allows this 1v TCXO to work, but with the new design R9's this "bug" was fixed and now they no longer work.
> 
> HOWEVER, you said these r9's work on the computer? Can you 100% confirm that? Because I would not have expected that to work, and if they do then there is potentially hope. I'm just not sure how they are working with the current TCXO

Yes, all HackRF from a new batch working fine with a PC. Already test it. 
I think if the quartz generator will be “bad” also HackRF will not work with a PC.

just look - all fake satellites signals are coming.
[image]

jLynx, 2023-07-25:
Yeah that is odd, maybe I'm wrong with it requiring the TCXO.

 Also you can confirm it's not using it by going to setting > radio, and it should say HackRF if it's using the 25mhz internal one or PortaPack or External if it's using the portapack one. Are you able to confirm what it says on the PortaPack?

jLynx, 2023-07-25:
Ok cool, so that confirms the HackRF is not using the TCXO on the PortaPack. No are able to remove the PP and test it with the computer again and see if it still works? (I'm not sure if it automatically runs in Hackrf mode with it removed, you may have to flash the HackRF firmware by itself to test)

Sats301, 2023-07-25:
> Ok cool, so that confirms the HackRF is not using the TCXO on the PortaPack. No are able to remove the PP and test it with the computer again and see if it still works? (I'm not sure if it automatically runs in Hackrf mode with it removed, you may have to flash the HackRF firmware by itself to test)
> 
> (I ask because Hackrf mode used diffrent code to PortaPack mode)

Ok, will test it ASAP, 40-60 min

jLynx, 2023-07-25:
The hackrf code is here btw if you do end up needing it https://github.com/greatscottgadgets/hackrf/releases

Sats301, 2023-07-25:
> The hackrf code is here btw if you do end up needing it https://github.com/greatscottgadgets/hackrf/releases

FW - done
[image]

Test the GPS Sim - no
[image]
[image]

I am completely not understand...
With a PP - working in HackRF mode - working normal.

bernd-herzog, 2023-07-25:
It looks like in hackrf mode it runs somewhat stable with the out of spec TCXO. The portapack firmware really doesn't like it and I'm not sure why that is.

Currently the portapack will not use any portapack TCXO if it runs on an r9 hackrf due to a bug. @Sats301 could try a (future) version that enables the use of the portapack TCXO and see whats happens. Maybe it is stable.

[portapack-h1_h2-mayhem_173_r9_clk_2.zip](https://github.com/eried/portapack-mayhem/files/12160857/portapack-h1_h2-mayhem_173_r9_clk_2.zip)
But maybe it is not stable, then it will turn on only every 20th power on try and when it turns on it will crash after starting an app. But you should be able to go to setting > radio. So give it a try.

Sats301, 2023-07-25:
> It looks like in hackrf mode it runs somewhat stable with the out of spec TCXO. The portapack firmware really doesn't like it and I'm not sure why that is.
>
> Currently the portapack will not use any [...]

Hello! Will check it ASAP. And return with feedback.

Sats301, 2023-07-25:
> @Sats301 , I also tried several times the EASTER.C8 And CATHEDRAL.C8 and the mobile detects the gps simul constellation but imagine that just with 20 secs (90 or 100 MB) no time to sync and detect any new position . I think you need good TXCO and fast SD card able to handle read 5.2 MB/sec. (2M6 x I,Q C8)
> 
> which MB size has the file that you have created ? How long time is the hole file to be replayed ? Are you replaying in loop or just once ? When you run with pc and HackRF mode, how long Does it takes to your mobile to get sync and decode a lat / long ?
> 
> If your file is longer than 20 secs and you can share it , I can also try it in an old HackRF r6 or older .
> 
> Thanks

No problem I will upload .C8 and parameters folder when I return to the lab and wil send the link. 
SD card with maximum speed, not more 32GB and also tested on 8GB.

Brumi-2021, 2023-07-26:
Hello  @Sats301 , no need to upload your created file.bin 
I already could create one as a testing and so far it works well from pc + hackrf , now pending to check from Portapack.  I will tell you soon

Brumi-2021, 2023-07-26:
Hi again @Sats301 ,  I could duplicate your problem .
With PC + Hackrf , it detects the new position in a few minutes (the first time, 20 secs definitively is not enough ).
With Portapack standalone  it should work  just copying the gpssim.c8 file there ,  but we have small bug , and it needs that you attach the related gpssim.txt    

It is just a TXT file with the same name as the gps_file_to_be_played.c8  , with same name.txt ,  with two lines ,
sample_rate=2600000
center_frequency=1575420000

You can try,  I already succeed  with TXCO and without TXCO  versions. 
We will ammend it soon .  but meanwhile , pls  confirm that workaround also works in your set up .

Cheers,

Sats301, 2023-07-26:
> 

YOOOOOO!!!!
Thank you! You save me!

Working fine!
Can you integrate this fix in 1.7.4?
[image]

Brumi-2021, 2023-07-26:
Glad to see that confirmation ! 
I already comment to @NotherNgineer and he is following it , and we will validate it together .
As soon as we fix it , we will integrate it . 
Our target is to solve it and integrate it in the next fw 1.7.4 . 
But as the problem is clear , and we have now a clear workaround ,  let's close that issue. (No need further investigation, but do not worry we will not forget it  ) .

jLynx, 2023-07-26:
I am working on integrating this fix already, hopefully it's not far away from release. There are just a few techinal challenges to get it to play nicely with everyone. 

CC: @Brumi-2021 leave it to me since I half way though doing it already

jLynx, 2023-07-26:
Also I'm keeping this issue open until the fix is implemented. That way we know it's still active

Sats301, 2023-07-26:
> Also I'm keeping this issue open until the fix is implemented. That way we know it's still active

Ok, Will wait for new release. Thank you so much!

Brumi-2021, 2023-07-26:
Hello @jLynx ,  
Sorry , but I already discuss and tested several versions from @NotherNgineer , and he already submittted a perfect PR .
I will add my evaluation commnets there , and I suggest to merge it .

jLynx, 2023-07-26:
Your changes are complelty different to what @bernd-herzog and I are talking about. So if your solves it that's great, but I also still need to implement these other changes

jLynx, 2023-07-26:
@Sats301 was the fix the firmware that @bernd-herzog sent you?

NotherNgineer, 2023-07-26:
@jLynx @bernd-herzog  I suspect it was the test firmware from bernd, since his initial post above showed the correct frequency & sample rate.

Brumi-2021, 2023-07-27:
Hi @Sats301 ,  

thanks to @NotherNgineer , yesterday , we just merged into latest nightly fw my yesterday proposed solution (when we do not attach the related .txt) 

But we had  two possible root causes about that issue , and we need to understand from you , if your confirmed solution  , 
         * was due to  this issue (already integrated,   set up correct parameters, when missing .txt file ) 
         *  or due to the successful new @bernd-herzog's  beta binary test (shared above) .  
                (If that is the case, we need to  know , in order to coordinate its validation / merge toward next fw 1.7.4.)

Waiting for your clarification about it , thanks,

Brumi-2021, 2023-07-27:
Ok , thanks for your confirmation ,
@jLynx was right , it was from the new binary .

@jLynx  , now you can close correctly that issue ,
Apologise for my confusion , we just fixed another minor second bug . 

Cheers

jLynx, 2023-07-27:
We wont close the issue until my PR is merged in https://github.com/eried/portapack-mayhem/pull/1313

Its almost ready. Just found a bug I am fixing

Sats301, 2023-07-27:
But last nightly release not fix this problem.

look:

1. with bernd-herzog FW
[image]

2. with nightly release n_230727
[image]

3. Radio settings scr 
[image]

Friends please check the difference…

jLynx, 2023-07-27:
@Sats301 Please try with the one I linked. Its not the same as the nightly. Its a custom built one

Sats301, 2023-07-27:
> @Sats301 Please try with the one I linked. Its not the same as the nightly. Its a custom built one https://github.com/eried/portapack-mayhem/files/12181665/portapack-h1_h2-mayhem.zip

Wait a second...

Sats301, 2023-07-27:
> @Sats301 Please try with the one I linked. Its not the same as the nightly. Its a custom built one
> https://github.com/eried/portapack-mayhem/files/12181665/portapack-h1_h2-mayhem.zip

Already test it. Results - not working.
[image]

Sats301, 2023-07-27:
[image]

jLynx, 2023-07-27:
Please upload that photo again, it seems it didnt attach correctly

jLynx, 2023-07-27:
Did you reboot in between enabling it?

Sats301, 2023-07-27:
> Did you reboot in between enabling it? (There is a bug I just found when rebooting it wont save the changes)

Yes. Set in Radio -> reboot -> test it.
Already test it. Results - not working.

[image]

[image]

jLynx, 2023-07-27:
ok can you please test without rebooting. So first disable it, save then enable it, save then test (without rebooting)

jLynx, 2023-07-27:
Scratch that. Please try this one here @Sats301 
[portapack-h1_h2-mayhem v2.zip](https://github.com/eried/portapack-mayhem/files/12182074/portapack-h1_h2-mayhem.v2.zip)

Sats301, 2023-07-27:
> ok can you please test without rebooting. So first disable it, save then enable it, save then test (without rebooting)

Checked - still no satellites

jLynx, 2023-07-27:
Please try with new v2 one I attached, after enabling it, try reboot and go into that menu and see if it shows 10mhz

Sats301, 2023-07-27:
> Scratch that. Please try this one here @Sats301
> [portapack-h1_h2-mayhem v2.zip](https://github.com/eried/portapack-mayhem/files/12182074/portapack-h1_h2-mayhem.v2.zip)

Tested - not working. 
Details: tested with settings external TCXO and without, with reboot and without.

Sats301, 2023-07-27:
> Please try with new v2 one I attached just above your last comment, after enabling it, try reboot and go into that menu and see if it shows 10mhz

Please wait a second

Sats301, 2023-07-27:
> Please try with new v2 one I attached just above your last comment, after enabling it, try reboot and go into that menu and see if it shows 10mhz

Yes, it change to 25, but still not working. 
[image]

if I trying to force to external TCXO, reference source comes to 10.

jLynx, 2023-07-28:
Actually @Sats301 I think that build still wont work. Are you able to please message me on Discord, It may be easier to get this working for you (My username is ``.jlynx``)

FrankVoice, 2023-08-05:
I was also having a problem with this GPS Sim not working. I did look all over the internet for an easy fix after reading this post to try to switch to internal clock. Can anyone please point me in right direction to use internal clock instead of external?  Also I noticed a sat showing up as 1575.420 and once time it was just 3 random letters but didn't get a screen shot. This was while I was watching which SATs Sim was uploading from Easter I believe it was.
[image]

Thank you

jLynx, 2023-08-05:
@FrankVoice still trying to work on a nice fix that will work for everyone, so you will just have to sit tight or use @bernd-herzog firmware he posted above for the moment

closed 2024-03-10 by gullradriel

smallwaye, 2024-03-16:
I'm having same issue, firmware and tools are most updated however gps sim on the pp doesn't work, it does work well in hack-rf mode with a computer, so how to get this fixed? thanks

junanxia, 2025-02-12:
> I'm having same issue, firmware and tools are most updated however gps sim on the pp doesn't work, it does work well in hack-rf mode with a computer, so how to get this fixed? tried enable or disable external txco but make no differences, i also tried to create or delete the TXT file sample_rate=2600000 center_frequency=1575420000 but same thing, anyone can help would be greatly appreciated,

I had the same problem. Did you solve it?