portapack-mayhem/mayhem-firmware #1334: Spectrum/Waterfall Screenshot doesn't match display

NotherNgineer, 2023-07-31:
A Screenshot of the Waterfall/Spectrum view doesn't match what appears on the screen.  This is possibly because Spectrum uses the rectangle "scroll" function.  Perhaps there's a way to fix this so that Screenshot files show what actually appeared on the display screen.

[image]

NotherNgineer, 2023-08-19:
Note that when a screenshot is taken with the DFU overlay displayed over the Spectrum display, the screen-shot image of the DFU overlay looks much better than it does on the actual screen.  (Scrolling messes up the DFU display on the actual screen.)  Hence, if code is added to fix the issue above, the DFU overlay should be taken into consideration.

Kiwax6z, 2023-09-09:
J'ai le même problème je n'arrive pas à résoudre

NotherNgineer, 2023-10-09:
If a Guru fault occurs when the screen is scrolled, the fault information (and any subsequent stack dumps) also displays incorrectly in the scrolled display area.

This part of the problem occurs without using Screenshot.