portapack-mayhem/mayhem-firmware #1343: Capture good: Playback at incorrect rate.

chris451gh, 2023-08-02:
----
Build: 1.74 release. Also began with 2023-08-81 build
(Please try the latest nightly release before submitting this. You can find the latest nightly version here: https://github.com/eried/portapack-mayhem/releases)

----

**Describe the bug**
A clear and concise description of what the bug is. 
Capture at 12k5 an upper sideband signal with voice on it.
Clean signal received.

Then play this signal back with Replay. 
The changes made to capture a nice clean signal have not been reversed in playback, 
so the playback is in chunks at high speed. a 22 second capture played back in 11.8 seconds, at the 12.5k capture rate. Filtering was good, snr was good, it was recorded with gain adusted as high as possible. Carrier suppression good. Signal type usb.
Signal type nfm: captured at 12.5k centered on signal. Also happens at 16k rate.
Good carrier suppression but the modulation sidebands are mixing making it unintelligable.
Playback speed: 2x. You can see this on the replay screen, 
look to the left of gain and it says 6 seconds. Then playback with loop enabled and time it, 3 seconds.
Did you forget to integrate 2x where it had been decimated 2x? Make the replay sample expand to fill the buffer, it does a fine job of playing it back at 2x.
Still has trouble with double sideband mixing. In most Gnu radios and SDRAngle configurations I played with, I put the NCO at 1/2 channel width or more, to get rid of the 0 crossover. There is usually a notch filter there to get rid of the carrier, but that kills the spectrum in the middle in addition to getting rid of the mixing carrier. For example 
Tune to 444.49 and place the nco at +10Khz (can be negative too 444.51 and -10khz)
The signal bandwith is 12.5k for nfm so the low edge begins at 444.50375 and the upper band edge is 444.5125 so the carrier area is evaded, thus no opposite sideband mixing is there in the playback.
For wider bandwidths just keep moving the NCO up. 
I tried a capture at 444.475 of a signal nfm at 444.500 and played it back, rate was set to 500k. 
The format was set to c16.
Its nearly perfect with no image and no carrier at 444.75. There was no signal on the image and perfect fidelity of the played back signal but 500k uses a lot of disk space.

 I dropped it down to 100k rate, recorded a signal at c8 and c16, played both chunks in a playlist.
 Very little difference when tuning to the original carrier 444.5 but there is an image.
 Tuned to 444.475. Signal at 444.500. Playback is at 444.5 and 444.45 there is an equal amplitude image.
I used a second tuner to look at the spectrum over a 64khz bandwidth and both sidebands appear about the 444.475 center frequency.
[image]

I wonder if the SSB app could be modified to accept wider bandwiths because it filters out the opposite sideband. In the SDRAngel app it has a capability of 24k (or +/- 24k in DSB mode.
That would help when recording a qpsk signal like p25, yet get rid of that opposite sideband.
It would allow recording qam signals that are starting to show up on the HF band.
Also some of the P25 simulcast systems have a lot of hetrodyne AM on them though they 
are trasmitting phase shift primarily.
They (different manufacturers) are also looking at wide bandwidths if you look at some of the experimantal licenses.

**To Reproduce**
Steps to reproduce the behavior:
1. Use capture at various rates
2. RePlay the files and look at the signal with a second SDR tuner like RTL-SDR or spectrum analyzer or another hackRF. 
3. Look for a mirror of the input signal at Fc-Signal offset where Signal offset is Fcenter -(Fsignal - Fcenter) Note for SSB or phase modulated signals USB becomes LSB and phase will flip from + to - 
4. Carrier suppression is good but the carrier is detectable in Replay
5. For any signal that is at the center of the tuner, the opposite sideband is mixed. 
6. at the low rate of 12.5k playback is clear as long as its only on one side of center and its playing back at 2x in 12.5k, 16k but  normal speed at 20k 25k and above. 
7. Due to opposite sideband the noise is 3db worse or more depending on if there is something there or not in Replay.
8. C8 capture rate saves space but I don't have a converter for it yet (to sdriq format) 

**Expected behavior**

A clear and concise description of what you expected to happen.
Opposite sideband is filtered out.
**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release:1.7.4
* Latest Nightly release:2023-08-01
* Previous working release:

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.
Notes...recent chages have improved SNR quite a bit and the carrier notch is working.
Gain control would be helpful for weak signals.

Brumi-2021, 2023-08-02:
Hi @chris451gh , thanks a lot for your check. 
In my previous evaluation  reports I already mentioned that so far in fw 1.7.4 ,
(1)  we implemented correct  files recording in the Capture in all BW options   (they are validated with inspectrum and gnuradio) 

(2) But I also mentioned  that the current fw  Replay could only replay correctly from BW 50k onwards . Now @kallanreed is working on it , and we are evaluating those new beta bin for solving 12k5, 16k, 20k, 25k . 
As soon  as we tested and validate it ,  we will merge it .

So as a summary

chris451gh, 2023-08-02:
Thank you very much I can see improvements already! It wasn't clear from the reading I've done but that explains it.

Brumi-2021, 2023-08-02:
Hi @chris451gh ,   
we just  validated now the beta binary of @kallanreed amending those Repaly issues (12k, 16k, 20k, 25k )  , 
soon we will propose to be merged, and you will see soon in the next nigthly . 

When you try to record  12k5 BW , you should really use a TX  very NBFM or  narrow AM  < than thatt the BW rec  spectrum , otherwise you will get  some distortion .  I still did not try  it with AM broadcastings , but using  NBFM with low deviation  , now with the new on going PR , Capture / Repay  12k5 16k , 20k looks good,  and 25k onwards really  very good .
But of course ,  we will appreciate as well ,  your evaluation and feedback .

Cheers.

Brumi-2021, 2023-08-07:
Hi @chris451gh ,  
We would appreciate if you can have a look again  on it,   using latest today's  nightly 7th of August or onwards.
I think now our Capure App is working well in all bit rate from 12k5 till 500khz  ,  (recently expanded also to 1Mhz and it should be OK selecting C8 , but when using C16 bits , it will depend a lot of microSD card performance)

Regarding the Replay , they should also work well, matching in time, with good spectrum scale...  , but it is true that at low bit rates , specially 12k5, 20k, 25k , the needed up-resampling interpolation inside  mayhem  are a basic and simple  one , not as good as the ones in gnuradio,... (but the recorded files quality are good, it is just a matter of interpolation complexity ) 

Once you could confirm it ,  pls. report here , and  if all good, pls. close that issue .

Cheers,

Brumi-2021, 2023-08-10:
Hi again @chris451gh ,

I have been testing more deeply this issue , and now I better understand the current fw performances.
With the latest  PR's , we have mananged to Capture correctly the file size and the seconds durations , (this is fine) .
But our  current solution is weak and not good enough in terms of anti aliasing filter and Image Reject filters. 

(1) Therefore ,  our current solution might be OK if you try to record a  single signal  (isolated , without adjacent channels ) 
That is why, when I was using isolated Signal Generator (with no adjacent channels in 1,5Mhz - 2Mhz distance) 
, all seems ok (with some slight background noise) 
[image]

(2) But when trying to record real MF , HF signals with close neightbour adjacent broadcastings (+-70-90Khz only )  , ...
our current Capture is only usefull from 250khz till  500Khz  (or 1Mhz recently added with fast enough  SD card and low write latency ) .  Clearly when trying to use BW = 100K  with close adjacent broadcastings , we are just having aliasing noise .

[image]
And even I improve the IR filter, we still have aliasing. 
[image]

Then I have to admit , that our current fw solution , has very good fw architecture basis , (with dynamic oversample rates) 
and now it really records the  expected bit rate based on the selected bandwith   , 
but clearly ,from RF Capture performance point of view,  it  still needs some more further improvement in the low bit rate recording part :  100k, 50k , 25k, 20k, 16k ,12k5  Capture App. 

Pending further investigations about it, but so far with real live adjacent channels, BW = >=250k and onwards are ok, 

Cheers.

closed 2023-08-14 by Brumi-2021