portapack-mayhem/mayhem-firmware #1344: Capture Pause

X3r0C00L, 2023-08-02:
Id like to see a feature in the capture app where you can set a squelch level and hit record and the recording will pause until the squelch is broken and then it will record and then stop .5-1 seconds after and wait for either another signal or to be stopped. 
Also just the ability to pause capturing manually would be great too.

sizzlik, 2023-08-02:
some sort of squelsh trigger to rec would be indeed a nice addition.

NotherNgineer, 2023-10-31:
Would it help if the "IQ Trim" app supported removing silent periods in the middle of a capture, versus just at the beginning and end?

gullradriel, 2024-03-10:
You can already do that in Recon.
That is, capture on squelch level activation, until X msecs of inactivity.

closed 2024-03-10 by gullradriel