portapack-mayhem/mayhem-firmware #1346: AFSK App still included?

sizzlik, 2023-08-03:
not a feature request but im courious about the AFSK App. The wiki says that the app doesnt work but just spits out random letters..wich it does. And by the wording of the article it seems nobody really knows who made the app and it must be inspected why it doesnt work. Im wondering why the app is still included then? Does the system rely on parts of it that work? Is it tied to something important? Or why isnt it removed to save some space and added back when it works?

sizzlik, 2023-08-27:
closing as nobody seems to know

closed 2023-08-27 by sizzlik

Kingdamienjl, 2024-11-18:
what is asfk?

gullradriel, 2024-11-18:
It's a radio protocol.
The app was removed because it was unfinished and not working.
Some other FSK apps are still here and working.
Stop thinking we are hiding things, we are not.
You are free to checkout the repository some years ago, use outdated firmware, and defunct apps. While it was a really good start, the old Havoc is not updated since years and was full of unfinished and buggy things.
We, on the other side, like clean things and are in need of firmware space, that is the reason why non working and or unfinished apps were removed, plus anyone can find them in the repository history.
Period.

dark-juju, 2025-02-23:
I have updated the [wiki on this subject](../wiki/AFSK) as it was quite bare.

I note here that the app is apparently removed, but it does seem to be in the nightly build at time of writing so I have assumed that this comment refers to the app being an external app and was previously part of the core firmware. Happy to correct if this is not the case - simply @ me.

gullradriel, 2025-02-23:
While your explanations might be okay, I'm not okay with the removal of the app status, that is "the app was beta and never worked like intended, and is not working".

I kept your modifications but added back the status on top of the page.

Thanks for your comprehension.

dark-juju, 2025-02-24:
Hey gullradriel - that makes sense.

I'm just happy to be contributing something to the project in anyway I can, and C++ is not my strongest language so sticking to the Wiki for the time being.