portapack-mayhem/mayhem-firmware #1353: scanner app support only one range entry !

alain00091, 2023-08-07:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly version here: https://github.com/eried/portapack-mayhem/releases)

release 23/08/06
----

**Describe the bug**
A clear and concise description of what the bug is. 

if you load a file containing 2 lines with 2 ranges, the first one is readed and scanned, NEVER the second line !

**To Reproduce**
Steps to reproduce the behavior:

Write those lines in a file and load it 
a=88000000,b=90000000,d=Test1
a=98100000,b=100000000,d=Test2

Only the Test1 is readed and scanned

**Expected behavior**
A clear and concise description of what you expected to happen.
Should read ALL the lines

**Affected versions**
I think since scanner app has been rewrited

Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release:
* Latest Nightly release:
* Previous working release:

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.

NotherNgineer, 2023-08-07:
This is by design, so changing it to "enhancement" request.

gullradriel, 2023-08-10:
Use the Recon app. It supports more than one range and allow list skip to get in whatever range you need.

closed 2023-08-16 by gullradriel