portapack-mayhem/mayhem-firmware #1354: scanner app Name of file is overwritted with Description if there is one

alain00091, 2023-08-07:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly version here: https://github.com/eried/portapack-mayhem/releases)
23/08/06
----

**Describe the bug**
A clear and concise description of what the bug is. 

if you start with Scanner.txt for example , without any description,  you can see the name of the loaded file
if you start a file with description , the name of the file is overwritted with description

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Tap on '....'
try it with/wo descriptions and you will see

**Expected behavior**
A clear and concise description of what you expected to happen.

we should see both description if any AND the file name

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release:
* Latest Nightly release:
* Previous working release:

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.

the problem is that the name of the file is written on the same line as description

NotherNgineer, 2023-08-07:
This is by design.  Changing to enhancement request.

gullradriel, 2023-08-10:
Use the Recon app.
It has separated line fields for each.

closed 2023-08-16 by gullradriel