portapack-mayhem/mayhem-firmware #1356: scanner app with one range overwrites search start and search end fields

alain00091, 2023-08-07:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly version here: https://github.com/eried/portapack-mayhem/releases)
23/08/06
----

**Describe the bug**
A clear and concise description of what the bug is. 

with a loaded file with one range line, it overwrites Manual Scan value ( start and end search fields)

**To Reproduce**
Steps to reproduce the behavior:

before put let's say 27 Mhz and 29Mhz in the manual scan fields
load file with a=88000000,b=90000000,d=Test and the manual scan fields are overwritten with 88.0 and 90.0

**Expected behavior**
A clear and concise description of what you expected to happen.

It should not overwrites manual scanner values, because  the manual scan is used for other frequency

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release:
* Latest Nightly release:
* Previous working release:

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.

NotherNgineer, 2023-08-07:
This is by design.  Changing to Enhancement request.

gullradriel, 2023-08-10:
Use the Recon app.
There is an option for allowing manual range overwrite or not.

closed 2023-08-16 by gullradriel