portapack-mayhem/mayhem-firmware #1357: scanner app with one range file does'nt show the frequency index and the max index

alain00091, 2023-08-07:
----

(Please try the latest nightly release before submitting this. You can find the latest nightly version here: https://github.com/eried/portapack-mayhem/releases)
23/08/06
----

**Describe the bug**
A clear and concise description of what the bug is. 

if you use a file with one range, it doesn't show anymore the index of the frequency scanned, and the max index as before
it shows --- / 0

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Tap on '....'
3. 

use a file with one line a=88000000,b=90000000,d=Test

**Expected behavior**
A clear and concise description of what you expected to happen.

We should see both index and max index

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable release:
* Latest Nightly release:
* Previous working release:

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.

NotherNgineer, 2023-08-07:
This is by design.  The index value only applies to individual frequencies in the Scan list (not search ranges).

gullradriel, 2023-08-16:
Same for Recon.
And I do not plan to support indexes of indexes.

closed 2023-08-16 by gullradriel